ardera/flutter-pi #360: gstreamer on x64 linux

bojidartonchev, 2023-09-12:
Starting this as a new thread since I am starting to investigate it. Playing video with the gstreamer player shows green screen. The issue occurs only on x64 linux.

https://github.com/ardera/flutter-pi/assets/12398245/bff662c2-43d2-4623-9769-8fc170eb146d

Best regards!

_Originally posted by @bojidartonchev in https://github.com/ardera/flutter-pi/issues/355#issuecomment-1709677108_

bojidartonchev, 2023-09-19:
I am not using `--pixelformat RGB565` since you published the fix for that, but the issue still occurs. I am using `--rotation 270` because my device monitor is in portrait mode. The video is in portrait mode too, 1080x1920. I have tried with landscape video without the `--rotation 270` argument, the result was the same.

[Here](https://www.metadata2go.com/result#j=2e4201a1-9b42-4bdc-b92e-1fd4d9c94a01&qr=true) you can see the metadata of the video I am trying to play.

bojidartonchev, 2023-10-23:
Hi @ardera, do you have any ideas that could help me in my investigation? And were you able to reproduce it on your x86 setup?

ardera, 2023-12-14:
I was not able to reproduce it, I'm afraid; I might have another idea on how to work around though

bojidartonchev, 2024-04-11:
Hi, did you have time check that issue, or maybe give me some hints I can work on?

bojidartonchev, 2025-07-16:
Hello,

Little update on that. I got it working with the following patch:

```
diff --git a/src/plugins/gstreamer_video_player/frame.c b/src/plugins/gstreamer_video_player/frame.c
index 66498ce..fc31607 100644
--- a/src/plugins/gstreamer_video_player/frame.c
+++ b/src/plugins/gstreamer_video_player/frame.c
@@ -359,7 +359,7 @@ UNUSED int dup_gst_buffer_range_as_dmabuf(struct gbm_device *gbm_device, GstBuff
         return -1;
     }
 
-    bo = gbm_bo_create(gbm_device, map_info.size, 1, GBM_FORMAT_R8, GBM_BO_USE_LINEAR);
+    bo = gbm_bo_create(gbm_device, 1080, 1920, GBM_FORMAT_ARGB8888, GBM_BO_USE_LINEAR);
     if (bo == NULL) {
         LOG_ERROR("Couldn't create GBM BO to copy video frame into.\n");
         goto fail_unmap_buffer;
@@ -415,7 +415,7 @@ UNUSED int dup_gst_memory_as_dmabuf(struct gbm_device *gbm_device, GstMemory *me
         return -1;
     }
 
-    bo = gbm_bo_create(gbm_device, map_info.size, 1, GBM_FORMAT_R8, GBM_BO_USE_LINEAR);
+    bo = gbm_bo_create(gbm_device, 1080, 1920, GBM_FORMAT_ARGB8888, GBM_BO_USE_LINEAR);
     if (bo == NULL) {
         LOG_ERROR("Couldn't create GBM BO to copy video frame into.\n");
         goto fail_unmap_buffer;
@@ -763,7 +763,7 @@ static EGLint egl_color_space_from_gst_info(const GstVideoInfo *info) {
     } else if (gst_video_colorimetry_matches(&GST_VIDEO_INFO_COLORIMETRY(info), GST_VIDEO_COLORIMETRY_BT2020)) {
         return EGL_ITU_REC2020_EXT;
     } else {
-        LOG_DEBUG("Unsupported video colorimetry: %s\n", gst_video_colorimetry_to_string(&GST_VIDEO_INFO_COLORIMETRY(info)));
+        //LOG_DEBUG("Unsupported video colorimetry: %s\n", gst_video_colorimetry_to_string(&GST_VIDEO_INFO_COLORIMETRY(info)));
         return EGL_NONE;
     }
 }
```

I am fine leaving it with a patch, but if you plan on fixing it, maybe this will help you.