ardera/flutter-pi #362: "Illegal instruction" error in release mode on Raspberry Pi 3

baharanchi, 2023-09-20:
I have a Raspberry Pi 3 B+ board running the official Raspbian bullseye arm64 OS. I could build and run the flutter sample app in debug mode and it worked fine. But after I built the app in release mode and tried to run it on the board, I got the `Illegal instruction` error and the app did not launch.

I used the following command in my Desktop PC to build the app:
```
flutterpi_tool build --release --arch=arm64 --cpu=pi3
```

And the following command on the Raspberry Pi to run the app (after deploying using `rsync`):
```
flutter-pi --release sample/
```
Which resulted in the `Illegal instruction` error.

Note that if I remove the `--release` flag from the above commands, everything works fine in debug mode.

ardera, 2023-09-20:
Okay, apparently Raspberry Pi 3 doesn't support cryptography extensions either.
Should be fixed on 3.13.4 once https://github.com/ardera/flutter-ci/actions/runs/6248966120 runs through.

Though you need to clear the flutter SDKs cache: `rm -rf <path-to-flutter-sdk>/bin/`

or maybe just do a complete `git clean -xfd` in the flutter sdk dir

baharanchi, 2023-09-20:
> Though you need to clear the flutter SDKs cache: `rm -rf <path-to-flutter-sdk>/bin/`

I think you meant `rm -rf <path-to-flutter-sdk>/bin/cache/`

> Okay, apparently Raspberry Pi 3 doesn't support cryptography extensions either. Should be fixed on 3.13.4 once https://github.com/ardera/flutter-ci/actions/runs/6248966120 runs through.

It is fixed. Thanks.

closed 2023-09-20 by baharanchi