ardera/flutter-pi #365: ApplicationException with Flutter 3.13.5

Rob-bb, 2023-09-23:
Hi, I am getting which I think is related to the version of flutter or dart I am using.  Flutter 3.13.5 

```
ERR : Failed to build flutterpi_tool:flutterpi_tool:
    | ../../snap/flutter/common/flutter/packages/flutter_tools/lib/src/vmservice.dart:1116:20: Error: The method 'getIsolatePauseEvent' isn't defined for the class 'VmService'.
    |  - 'VmService' is from 'package:vm_service/src/vm_service.dart' ('../../.pub-cache/hosted/pub.dev/vm_service-11.3.0/lib/src/vm_service.dart').
    | Try correcting the name to the name of an existing method, or defining a method named 'getIsolatePauseEvent'.
    |     return service.getIsolatePauseEvent(isolateId)
    |                    ^^^^^^^^^^^^^^^^^^^^
FINE: Exception type: ApplicationException
```

ardera, 2023-09-24:
Can you try updating flutterpi_tool with `flutter pub global activate flutterpi_tool`?

Rob-bb, 2023-09-25:
Yea, that seems to have done the trick, it works fine now, 😃

closed 2023-10-03 by ardera