ardera/flutter-pi #366: NoSQL database?

BigGitWorld, 2023-09-26:
Hello.
Can anyone suggest a good dart NoSQL database with support for amr64 architecture?

pieter-scholtz, 2023-09-27:
https://github.com/isar/isar/issues/876 , seems one can easily make isar work with flutter-pi

BigGitWorld, 2023-09-27:
> [isar/isar#876](https://github.com/isar/isar/issues/876) , seems one can easily make isar work with flutter-pi

How? Do I have to build the isar in a Raspberry-pi?

pieter-scholtz, 2023-09-28:
Well , building it on the raspberry pi is an option , but I believe there is already a way build it since the author mentions it in the issue that he can build ar linux arm64 binary. It seems to be part of the build scripts ( https://github.com/isar/isar/blob/main/tool/build_linux.sh )

BigGitWorld, 2023-09-30:
> @BigGitWorld You can try this binary , I haven't tested it yet [binary.zip](https://github.com/ardera/flutter-pi/files/12763409/binary.zip) it is for the latest stable release , `isar-3.1.0-1`

Dear @pieter-scholtz 
Thank you for your help.
I added your ``libisar.so`` file to the Raspberry-pi (``/usr/local/lib``) and updated Shared Library Cache Using ``sudo ldconfig``, but again got the same error:

```
IsarError: Unsupported processor architecture "linux_arm64". Please open an issue on GitHub to request it.
```

pieter-scholtz, 2023-10-02:
@BigGitWorld Are you sure that the library from `/usr/local/lib` is the one that is getting used ? , perhaps you can point it to the binary using https://pub.dev/documentation/isar/latest/isar/Isar/initializeIsarCore.html

pieter-scholtz, 2023-10-02:
@BigGitWorld https://github.com/isar/isar/blob/6643d064abf22606b6c6a741ea873e4781115ef4/packages/isar/lib/src/native/isar_core.dart#L207
You probably need to add a new line in between ln207 and 208 and put this : `case Abi.linuxArm64:`

BigGitWorld, 2023-10-04:
> @BigGitWorld https://github.com/isar/isar/blob/6643d064abf22606b6c6a741ea873e4781115ef4/packages/isar/lib/src/native/isar_core.dart#L207 You could try adding a new line in between ln207 and 208 and put this : `case Abi.linuxArm64:`

I did same and got the following error:
```
Invalid argument(s): Failed to load dynamic library '/home/pi/libisar.so': /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/pi/libisar.so)
```
How did you build ``libisar.so``?can you please write that command here?

closed 2023-10-04 by BigGitWorld