ardera/flutter-pi #369: flutter_secure_storage MissingPluginException

INDAPP, 2023-11-02:
I'm trying to run an app with `flutter_secure_storage` plugin.

When I try to run the app I receive the following error:

```sh
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method read on channel plugins.it_nomads.com/flutter_secure_storage)
```

The application starts, but I cannot proceed without a secure storage.

The application works fine on other platforms like macOS or ubuntu.

I already tried to install `libsecret-1-0` from apt-get, but the problem seems to be not related.

In the readme file I read that I "have to build the platform side of the plugins"... but how?

ardera, 2024-01-05:
Yeah you're right, flutter_secure_storage is not supported on flutter-pi and you'd need to write some C code to make it work.

Though this package might work: https://pub.dev/packages/amplify_secure_storage_dart

vanlooverenkoen, 2024-11-07:
I have created a ticket at flutter_secure_storage here (to track support for flutter pi) 
https://github.com/mogol/flutter_secure_storage/issues/819

vanlooverenkoen, 2024-11-07:
@ardera the amplify_secure_storage_dart always returns `null` so to me it seems like this one is also not working.

Anyone who has got this working?

eximius313, 2026-11-12:
@vanlooverenkoen, 
I'm also using https://pub.dev/packages/flutter_secure_storage

I see that https://github.com/juliansteenbakker/flutter_secure_storage/issues/819 is completed, but I still see:
```
MissingPluginException(No implementation found for method read on channel plugins.it_nomads.com/flutter_secure_storage)
```
Have you managed to make it work?