ardera/flutter-pi #371: Error: Could not load flutter engine from any location

rsov, 2023-11-29:
## Building

Steps on linux (WSL) system:
```
git clone https://github.com/flutter/gallery.git flutter_gallery
cd flutter_gallery
git checkout d77920b4ced4a105ad35659fbe3958800d418fb9
flutter pub get
flutterpi_tool build --release --cpu=pi4
scp -r ./build/flutter_assets pi@raspberrypi:/home/pi/flutter_gallery
```

Get warning:
```
2023/11/29 14:04:38.899981 system_key.go:129: cannot determine nfs usage in generateSystemKey: cannot parse mountinfo: incorrect number of tail fields, expected 3 but found 4
2023/11/29 14:04:38.902915 cmd_run.go:1046: WARNING: cannot create user data directory: cannot determine SELinux status: failed to obtain SELinux mount path: incorrect number of tail fields, expected 3 but found 4
```

## Running

Steps on pi:
```
flutter-pi --release ~/flutter_gallery
```
Get error:
```
flutter-pi.c: Error: Could not load engine from any location. Make sure you have installed the engine binaries.
```

The `libflutter_engine.so` is included inside the `flutter_gallery`  directory

ardera, 2023-12-14:
If the engine can't be loaded, could also be it's for the wrong architecture, or otherwise not executable (somehow).

Configuring and building flutter-pi with `-DCMAKE_BUILD_TYPE=Debug` should give more information. What does `file ~/flutter_gallery` say for you?

rsov, 2023-12-19:
## cmake output
```
pi@raspberrypi:~/flutter-pi/build $ cmake .. -DCMAKE_BUILD_TYPE=Debug
-- The C compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Debug
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'libdrm'
--   Found libdrm, version 2.4.114
-- Checking for module 'gbm'
--   Found gbm, version 23.2.1-1~bpo12+rpt2
-- Checking for module 'libsystemd'
--   Found libsystemd, version 252
-- Checking for module 'libinput'
--   Found libinput, version 1.22.1
-- Checking for module 'xkbcommon'
--   Found xkbcommon, version 1.5.0
-- Checking for module 'libudev'
--   Found libudev, version 252
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Checking for module 'egl'
--   Found egl, version 1.5
-- Checking for module 'glesv2'
--   Found glesv2, version 3.2
-- EGL/GLES support ....... ON
-- Lint EGL headers ....... OFF
-- Checking for module 'vulkan'
--   Package 'vulkan', required by 'virtual:world', not found
-- Vulkan support ......... OFF
-- Filesystem Layout ...... default
-- Checking for module 'libseat'
--   Package 'libseat', required by 'virtual:world', not found
libseat was not found. flutter-pi will be built without session switching support.
-- Session switching ...... OFF
-- Checking for module 'gstreamer-1.0'
--   Found gstreamer-1.0, version 1.22.0
-- Checking for module 'gstreamer-plugins-base-1.0'
--   Found gstreamer-plugins-base-1.0, version 1.22.0
-- Checking for module 'gstreamer-app-1.0'
--   Found gstreamer-app-1.0, version 1.22.0
-- Checking for module 'gstreamer-allocators-1.0'
--   Found gstreamer-allocators-1.0, version 1.22.0
-- Checking for module 'gstreamer-video-1.0'
--   Found gstreamer-video-1.0, version 1.22.0
-- Checking for module 'gstreamer-audio-1.0'
--   Found gstreamer-audio-1.0, version 1.22.0
-- Performing Test COMPILER_SUPPORTS_MACRO_PREFIX_MAP
-- Performing Test COMPILER_SUPPORTS_MACRO_PREFIX_MAP - Success
-- IPO/LTO ................ OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/flutter-pi/build
```

## make output
```
pi@raspberrypi:~/flutter-pi/build $ make -j`nproc`
[  8%] Building C object CMakeFiles/flutterpi_module.dir/src/pluginregistry.c.o
[  8%] Building C object CMakeFiles/flutterpi_module.dir/src/flutter-pi.c.o
[  8%] Building C object CMakeFiles/flutterpi_module.dir/src/platformchannel.c.o
[ 11%] Building C object CMakeFiles/flutterpi_module.dir/src/texture_registry.c.o
[ 14%] Building C object CMakeFiles/flutterpi_module.dir/src/modesetting.c.o
[ 17%] Building C object CMakeFiles/flutterpi_module.dir/src/util/collection.c.o
[... 117 lines cut ...]
    CPU min MHz:         600.0000
    BogoMIPS:            38.40
    Flags:               fp asimd evtstrm crc32 cpuid
Caches (sum of all):
  L1d:                   128 KiB (4 instances)
  L1i:                   128 KiB (4 instances)
  L2:                    512 KiB (1 instance)
Vulnerabilities:
  Gather data sampling:  Not affected
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Not affected
  Retbleed:              Not affected
  Spec rstack overflow:  Not affected
  Spec store bypass:     Not affected
  Spectre v1:            Mitigation; __user pointer sanitization
  Spectre v2:            Not affected
  Srbds:                 Not affected
  Tsx async abort:       Not affected
  
pi@raspberrypi:~/my_apps_flutter_assets $ uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

rsov, 2023-12-19:
Tried building with arm64 `flutterpi_tool build --release --cpu=pi3 --arch=arm64`

Got a different error
```
pi@raspberrypi:~ $ flutter-pi --release /home/pi/my_apps_flutter_assets/
==============Locale==============
Flutter locale:
  default: en
  locales: en en.UTF-8 en_GB en.UTF-8 en.UTF-8 en_GB en.UTF-8
===================================
===================================
EGL information:
  version: 1.4
  vendor: Mesa Project
  client extensions: EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_EXT_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless
  display extensions: EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display
===================================
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 23.2.1-1~bpo12+rpt2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_generate_mipmap_sRGB GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_debug_label GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_EXT_texture_compression_s3tc_srgb GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order GL_MESA_bgra "
===================================
window.c: WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
Use the `-d` commandline parameter to specify the physical dimensions of your display.
display mode:
  resolution: 1920 x 1080
  refresh rate: 60.000000Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
  pixel format: (any)
pluginregistry.c: Initialized plugins: services, text input, raw keyboard plugin, gstreamer video_player, audioplayers,
window.c: Flutter requested render surface before supplying surface dimensions.
gl_renderer.c: Choosing EGL config with pixel format ARGB 8:8:8:8...
Illegal instruction
```

mancvso, 2025-02-16:
It happened to me when switching screen drivers (dto), solved switching drivers again.
In my case
faulty: pitft35-resistive
worked: piscreen

ardera, 2025-03-20:
> iam getting same error was there a solution to this

This should be fixed now, Pi 3 & 4 don't support cryptography instructions and CI was incorrectly building with these.
If it's not fixed, you can provide a core dump, so I can see what's going on.

eximius313, 2026-10-31:
@ardera, I'm using pi4
I've done:
```
flutter pub global activate --source git https://github.com/ardera/flutterpi_tool.git
flutterpi_tool build --release --cpu=pi4
```
Although I don't have `./build/flutter_assets/` I have `./build/flutter-pi/pi4/` with
```
app.so
flutter-pi
icudtl.dat
native_assets
OPEN_SOURCE_LICENSES.txt
AssetManifest.bin 
FontManifest.json 
libflutter_engine.so 
NativeAssetsManifest.json  
shaders
assets
fonts
linux
NOTICES.Z
version.json
```
so I did 
```
scp -r ./build/flutter-pi/ pi@raspberrypi:/home/pi/flutter-pi
```
But when I do
```
flutter-pi --release flutter-pi/pi4/
```
I get
```
flutter-pi.c: Error: Could not load flutter engine from any location. Make sure you have installed the engine binaries.
```
I'm a bit confused, because I used "new method" and I see that `Install the engine-binaries (not required anymore, except when using the old method to build the app bundle, see below)`

eximius313, 2026-11-01:
@ardera, in [Changelog](https://pub.dev/packages/flutterpi_tool/changelog) I see
```
0.12.0 - 2026-07-22 
flutter 3.44.x compatibility (thanks to @jpohhhh)
```

I'm using 
```
Flutter 3.47.2 • channel stable 
```
so perhaps this is the problem?
If so, then the error message:
```
Make sure you have installed the engine binaries.
```

is a bit misleading. I'd expect that the `flutterpi_tool` would display the message `The tool is not yet compatibile with your version of Flutter` while building.

eximius313, 2026-11-04:
@ardera, I've just found out, that there is already commit titled "[add support for flutter 3.47](https://github.com/ardera/flutterpi_tool/commit/248d79914b84ce89b44435ad8f104e81a37e1ffc)" in master, so now I'm confused (as I've installed `flutterpi_tool` using `flutter pub global activate --source git https://github.com/ardera/flutterpi_tool.git`)

Do you have an ideas what could be the source of `Could not load flutter engine from any location.`?

eximius313, 2026-11-12:
@ardera, I'd love to help with this issue. Any hints on where could I start?

ardera, 2026-11-12:
@eximius313 sorry for the delayed reply, this is a different issue than the original one

it might be a permissions issue in your case, you can try adding your device as a flutterpi_tool device (see here: https://github.com/ardera/flutterpi_tool#examples) and letting that copy over for you. or just try running `chmod +x` on the `libflutter_engine.so`.

the "make sure you have installed the engine binaries" error message is outdated, you're right, that's not required anymore

if that still fails:
- the flutterpi_tool output with `--verbose` flag would be helpful
- the output of running flutter-pi manually with env var `LD_DEBUG=all`

eximius313, 2026-11-12:
@ardera, thank you for the reply.
1. Is `flutterpi_tool devices add` required also when I don't use `flutterpi_tool run -d pi5`, but instead I scp the build, then I ssh to it, and I run the project manually? If so, It would be great to add clarifcation to the docs (as I was sure that adding device is only necessary when one want to use run command)
2.  I had:
```
ls -la ./flutter-pi/pi4/libflutter_engine.so
-rw-r--r-- 1 pi pi 19699504 Oct 31 19:20 ./flutter-pi/pi4/libflutter_engine.so
```
But even when changed to:
```
-rwxr-xr-x  1 pi pi 19699504 Oct 31 19:20 libflutter_engine.so
```
it failed

`LD_DEBUG=all` log: [debug.log.gz](https://github.com/user-attachments/files/34768866/debug.log.gz)

eximius313, 2026-11-12:
Running `flutterpi_tool run -d pi4` 
shown:
```
Exception: Make sure all required runtime dependencies are installed on the device.
For example, for targets runned debian-based distros, you can execute:
  ssh -o BatchMode=yes pi@pi4 'sudo sh -c "apt-get update && apt-get install -y libdrm2 libgbm1 libsystemd0 libinput10 libxkbcommon0 libudev1 libegl1 libgles2 libvulkan1 libgstreamer1.0-0 libgstreamer-plugins-base1.0-0 libglib2.0-0"'
Error launching application on pi4 (wireless)
```

I did that, and the output was:
```
The following NEW packages will be installed:
  gstreamer1.0-plugins-base libcdparanoia0 libgstreamer-plugins-base1.0-0 libgstreamer1.0-0 libopus0 liborc-0.4-0t64 libtheoradec1 libtheoraenc1 libvisual-0.4-0 libvorbisenc2
```
But I'm pretty sure that I've installed all dependencies mentioned here: https://github.com/ardera/flutter-pi
```
sudo apt install cmake libgl1-mesa-dev libgles2-mesa-dev libegl1-mesa-dev libdrm-dev libgbm-dev ttf-mscorefonts-installer fontconfig libsystemd-dev libinput-dev libudev-dev  libxkbcommon-dev
```

Anyways, after installing theese dependencies, the error was the same for `flutter-pi --release` **but** `flutterpi_tool run -d pi4` sucessfully run the application.

Where does `flutterpi_tool run -d pi4` install the app, so I could check what's different?

ardera, 2026-11-12:
> 1. Is flutterpi_tool devices add required also when I don't use flutterpi_tool run -d pi5, but instead I scp the build, then I ssh to it, and I run the project manually? If so, It would be great to [...]

no, you had it right, it's only required for flutterpi_tool run

> Where does flutterpi_tool run -d pi4 install the app, so I could check what's different?

here it is: https://github.com/ardera/flutterpi_tool/blob/0d02f0a62f3ed6f00b4b59b1c0f66140eb78cf84/lib/src/devices/flutterpi_ssh/device.dart#L231-L283

it's basically `scp` followed by `chmod +x` :)

if it says it can't find the engine library, almost always something is wrong with the library. maybe it's not executable, maybe it's the wrong CPU architecture, etc

eximius313, 2026-11-12:
> here it is: https://github.com/ardera/flutterpi_tool/blob/0d02f0a62f3ed6f00b4b59b1c0f66140eb78cf84/lib/src/devices/flutterpi_ssh/device.dart#L231-L283

oh, so basically `/tmp` ;)

I see few differences.
```
flutterpi_tool build --release --cpu=pi4
scp -r ./build/flutter-pi/ pi@raspberrypi:/home/pi/flutter-pi
```
gives:
```
ls -la .
total 29132
drwxr-xr-x 7 pi pi     4096 Oct 31 19:22 .
drwx------ 9 pi pi     4096 Nov 12 21:41 ..
-rw-r--r-- 1 pi pi  8438344 Oct 31 19:22 app.so
-rw-r--r-- 1 pi pi      171 Oct 31 19:22 AssetManifest.bin
drwxr-xr-x 2 pi pi     4096 Oct 31 19:22 assets
-rw-r--r-- 1 pi pi   376352 Oct 31 19:20 flutter-pi
-rw-r--r-- 1 pi pi       82 Oct 31 19:22 FontManifest.json
drwxr-xr-x 2 pi pi     4096 Oct 31 19:22 fonts
-rw-r--r-- 1 pi pi   864880 Oct 31 19:20 icudtl.dat
-rw-r--r-- 1 pi pi       32 Oct 31 19:22 .last_build_id
-rwxr-xr-x 1 pi pi 19699504 Oct 31 19:20 libflutter_engine.so
drwxr-xr-x 3 pi pi     4096 Oct 31 19:20 linux
drwxr-xr-x 3 pi pi     4096 Oct 31 19:22 native_assets
-rw-r--r-- 1 pi pi       45 Oct 31 19:22 NativeAssetsManifest.json
-rw-r--r-- 1 pi pi   112002 Oct 31 19:22 NOTICES.Z
-rw-r--r-- 1 pi pi   278391 Oct 31 19:22 OPEN_SOURCE_LICENSES.txt
drwxr-xr-x 2 pi pi     4096 Oct 31 19:22 shaders
-rw-r--r-- 1 pi pi       75 Oct 31 19:22 version.json
```
while `flutterpi_tool run -d pi4` gives:
```
total 141384
drwx---rwx 6 pi pi      340 Nov 12 23:33 .
drwxrwxrwt 9 root  root       180 Nov 12 23:32 ..
-rw-rw-r-- 1 pi pi      171 Nov 12 23:32 AssetManifest.bin
drwx---rwx 2 pi pi       80 Nov 12 23:32 assets
-rwxrwxr-x 1 pi pi  1164768 Nov 12 23:32 flutter-pi
-rw-rw-r-- 1 pi pi       82 Nov 12 23:32 FontManifest.json
drwx---rwx 2 pi pi       60 Nov 12 23:32 fonts
-rw-rw-r-- 1 pi pi   864880 Nov 12 23:32 icudtl.dat
-rw-rw-r-- 1 pi pi 93697624 Nov 12 23:32 kernel_blob.bin
-rw-rw-r-- 1 pi pi       32 Nov 12 23:32 .last_build_id
-rwxrwxr-x 1 pi pi 48623880 Nov 12 23:33 libflutter_engine.so
drwx---rwx 3 pi pi       60 Nov 12 23:33 native_assets
-rw-rw-r-- 1 pi pi       45 Nov 12 23:33 NativeAssetsManifest.json
-rw-rw-r-- 1 pi pi   112147 Nov 12 23:33 NOTICES.Z
-rw-rw-r-- 1 pi pi   278391 Nov 12 23:33 OPEN_SOURCE_LICENSES.txt
drwx---rwx 2 pi pi       80 Nov 12 23:33 shaders
-rw-rw-r-- 1 pi pi       75 Nov 12 23:33 version.json
```

Few notable changes:
1. Former has `app.so` while the latter does not
2. size of `libflutter_engine.so` in former is `19699504` while in latter is `48623880`
3. Latter has also `executable` directories. I'm not sure if it's necessary
4. Also `flutter-pi` is not executable in former while it is in latter
5. Former does not have `kernel_blob.bin` while the latter has

eximius313, 2026-11-12:
I've noticed, that `flutterpi_tool run -d pi4` created `build/flutter-pi/aarch64-generic` next to `build/flutter-pi/pi4`, and when I ran `flutterpi_tool build --release --arch=arm64 --cpu=pi4` (instead of just `flutterpi_tool build --release --cpu=pi4`), it downloaded additional:
```
Downloading engine-aarch64-generic-release.tar.xz...             2,110ms
Downloading engine-pi4-64-release.tar.xz...                      1,869ms
```
I've scp -r this `./flutter-pi/pi4-64` and it worked just fine, so perhaps despite saying that:
```
          [pi4]                                              Use a Raspberry Pi 4 tuned engine. Compatible with arm and
                                                             arm64. (-mcpu=cortex-a72+nocrypto -mtune=cortex-a72)
```
`pi4` is not compatibile with arm64?

Also, no `chmod +x` was necessary:
```
ls -la pi4-64/
total 30228
drwxr-xr-x 7 pi pi     4096 Nov 13 00:24 .
drwx------ 9 pi pi     4096 Nov 13 00:22 ..
-rw-r--r-- 1 pi pi  8258440 Nov 13 00:22 app.so
-rw-r--r-- 1 pi pi      171 Nov 13 00:22 AssetManifest.bin
drwxr-xr-x 2 pi pi     4096 Nov 13 00:22 assets
drwx------ 3 pi pi     4096 Nov 13 00:24 flutter_engine
-rw-r--r-- 1 pi pi   376456 Nov 13 00:22 flutter-pi
-rw-r--r-- 1 pi pi       82 Nov 13 00:22 FontManifest.json
drwxr-xr-x 2 pi pi     4096 Nov 13 00:22 fonts
-rw-r--r-- 1 pi pi   864880 Nov 13 00:22 icudtl.dat
-rw-r--r-- 1 pi pi       32 Nov 13 00:22 .last_build_id
-rw-r--r-- 1 pi pi 21003672 Nov 13 00:22 libflutter_engine.so
drwxr-xr-x 3 pi pi     4096 Nov 13 00:22 native_assets
-rw-r--r-- 1 pi pi       45 Nov 13 00:22 NativeAssetsManifest.json
-rw-r--r-- 1 pi pi   112147 Nov 13 00:22 NOTICES.Z
-rw-r--r-- 1 pi pi   278391 Nov 13 00:22 OPEN_SOURCE_LICENSES.txt
drwxr-xr-x 2 pi pi     4096 Nov 13 00:22 shaders
-rw-r--r-- 1 pi pi       75 Nov 13 00:22 version.json
```

Thanks for help @ardera !