ardera/flutter-pi #375: Pre-built flutter-pi binaries?

kerberjg, 2024-01-01:
Would it be possible to pre-build flutter-pi itself outside of a Raspberry Pi? Such as in an ARM64 container/VM?

We're working on a project that doesn't allow us to install the whole toolchain on the RPi (space constraints), so I'm wondering if flutter-pi itself can be built "elsewhere", like the engine binaries and the app itself are.

I'm going to experiment a bit with doing so in Docker these days and report back the findings, but I'm wondering if there are any specific limitations that would hinder this

kerberjg, 2024-01-01:
I confirm I was able to build flutter-pi in an `aarch64` Docker container of `debian:bullseye-slim`.

The only roadblock was that I wasn't able to install the fonts package, but to my understanding it's a requirement for runtime, not build? (Please confirm)

I'll try later to see if runs on an actual RasPi

ardera, 2024-01-02:
Hey, yes that works, that's what I'm doing myself for some devices.

And you're right, fonts is a runtime requirement and you don't need it for building. Basically you only need the `-dev` packages for building.

kerberjg, 2024-01-02:
Once I’m able to confirm my build procedure works, would you mind if I contributed a small GHA workflow for CI builds?

I feel like this can be very useful for constrained environments and just overall useful to get people started more quickly 😊

ardera, 2024-01-02:
> Once I’m able to confirm my build procedure works, would you mind if I contributed a small GHA workflow for CI builds?
> 
> I feel like this can be very useful for constrained environments and just overall useful to get people started more quickly 😊

Yes of course! Also think it'd be useful.

kerberjg, 2024-01-03:
Alright! I was able to confirm it working, when compiling on M1+Docker I get the same output hash as compiling it on the Pi, I'm taking this as a sign that it works 😊 (props to @marcociarrocca for the assist)

I'll have the PR up within the day