ardera/flutter-pi #376: Issue with GST Video Player on RPI3B

cfsbhawkins, 2024-01-02:
I am running the latest flutter-pi code, on the latest version of Raspberry PI OS, when I try and start a video playback using the Pipeline commands, I get the following error in the console on the PI.

```shell
Linux videoplayer 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Jan  2 21:18:27 2024 from 10.4.223.137
pi@videoplayer:~ $ flutter-pi --release flutter_assets
plugins/gstreamer_video_player/plugin.c: Couldn't create gstreamer video player.
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(nativeerror, Input/output error, 5, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:322)
<asynchronous suspension>
#2      FlutterpiVideoPlayer.create (package:flutterpi_gstreamer_video_player/src/platform.dart:78)
<asynchronous suspension>
#3      VideoPlayerController.initialize (package:video_player/video_player.dart:438)
<asynchronous suspension>
#4      ChewieController._initialize (package:chewie/src/chewie_player.dart:557)
<asynchronous suspension>
``` 

I have tried both the videotestsrc and a playbin to a folder a files both fail with the same error. I am using the example from pub.dev to make the video player widget.

```dart
super.initState();
    _controller = FlutterpiVideoPlayerController.withGstreamerPipeline(
      'videotestsrc ! queue ! appsink name="sink"',
    );
``` 

Video Test SRC should just render color bars on the screen.

ardera, 2024-01-05:
Thanks for reporting, should be fixed now: 7dc5f999a72d5bd00f0e383e01f247c1a4761602
In my own tests it worked, can you try if it works for you?

cfsbhawkins, 2024-01-05:
> Thanks for reporting, should be fixed now: [7dc5f99](https://github.com/ardera/flutter-pi/commit/7dc5f999a72d5bd00f0e383e01f247c1a4761602) In my own tests it worked, can you try if it works for you?

Can confirm it now works, thanks!

closed 2024-01-05 by cfsbhawkins