ardera/flutter-pi #378: Failed to build bundle with flutterpi_tool

quanggdungg0609, 2024-01-14:
i have the problem when build the demo app, here is my errors:
```
parallels@ubuntu-linux-22-04-desktop:~/hello$ flutterpi_tool build --cpu=pi4 --release
Downloading engine-armv7-generic-release.tar.xz...               2,943ms
Downloading engine-pi4-release.tar.xz...                         2,645ms
Downloading universal.tar.xz...                                  2,438ms
Target aot_elf_release failed: ProcessException: Failed to find
"/home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/flutterp
i-gen-snapshot-linux-arm64-armv7-generic-release/gen_snapshot" in the search
path.
  Command:
  /home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/flutter
  pi-gen-snapshot-linux-arm64-armv7-generic-release/gen_snapshot
#0      LocalProcessManager.start (package:process/src/interface/local_process_manager.dart:56:7)
#1      ErrorHandlingProcessManager.start.<anonymous closure> (package:flutter_tools/src/base/error_handling_io.dart:689:24)
#2      _run (package:flutter_tools/src/base/error_handling_io.dart:559:20)
#3      ErrorHandlingProcessManager.start (package:flutter_tools/src/base/error_handling_io.dart:688:12)
#4      _DefaultProcessUtils.start (package:flutter_tools/src/base/process.dart:430:28)
#5      _DefaultProcessUtils.stream (package:flutter_tools/src/base/process.dart:450:35)
#6      GenSnapshot.run (package:flutter_tools/src/base/build.dart:75:26)
#7      AOTSnapshotter.build (package:flutter_tools/src/base/build.dart:222:56)
#8      AotElfBase.build (package:flutter_tools/src/build_system/targets/common.dart:318:52)
#9      _BuildInstance._invokeInternal (package:flutter_tools/src/build_system/build_system.dart:853:27)
<asynchronous suspension>
#10     Future.wait.<anonymous closure> (dart:async/future.dart:523:21)
<asynchronous suspension>
#11     _BuildInstance.invokeTarget (package:flutter_tools/src/build_system/build_system.dart:791:32)
<asynchronous suspension>
#12     Future.wait.<anonymous closure> (dart:async/future.dart:523:21)
<asynchronous suspension>
#13     _BuildInstance.invokeTarget (package:flutter_tools/src/build_system/build_system.dart:791:32)
<asynchronous suspension>
#14     FlutterBuildSystem.build (package:flutter_tools/src/build_system/build_system.dart:620:16)
<asynchronous suspension>
#15     buildFlutterpiBundle (package:flutterpi_tool/src/build_bundle.dart:366:20)
<asynchronous suspension>
#16     BuildCommand.run.<anonymous closure> (package:flutterpi_tool/src/build_bundle.dart:682:11)
<asynchronous suspension>
#17     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#18     BuildCommand.run (package:flutterpi_tool/src/build_bundle.dart:646:5)
<asynchronous suspension>
#19     CommandRunner.runCommand (package:args/command_runner.dart:212:13)
<asynchronous suspension>
#20     main (package:flutterpi_tool/src/build_bundle.dart:789:5)
<asynchronous suspension>
#21     main (file:///home/parallels/.pub-cache/hosted/pub.dev/flutterpi_tool-0.2.1/bin/flutterpi_tool.dart:4:3)
<asynchronous suspension>

Building Flutter-Pi bundle...                                           
Failed to build bundle.

```
Can someone help me find a solution.

IceIce1030, 2024-02-06:
Has this issue been resolved?

ardera, 2024-02-20:
@IceIce1030 yes (@quanggdungg0609 also reported it on discord, and we talked there)

Linux arm64 is not supported as a host platform right now. MacOS arm64 is supported though, in the above case he just switched to that.

payam-zahedi, 2024-02-20:
The other question is how can connect my app to devtools?

mrnetforge, 2024-02-27:
> flutterpi_tool doesn't run apps (yet), so you need to use flutter custom-devices for that

I configure my Raspberry PI  device, and now it's in `flutter devices`. However when i create `lanuch.json` inside Vscode:

````
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "pi_attendance_app",
            "request": "launch",
            "type": "dart"
        },
        {
            "name": "pi_attendance_app (profile mode)",
            "request": "launch",
            "type": "dart",
            "flutterMode": "profile"
        },
        {
            "name": "pi_attendance_app (release mode)",
            "request": "launch",
            "type": "dart",
            "flutterMode": "release"
        }
    ]
}
````

Select device from list and then press Run i have:

```
Launching lib/main.dart on Raspberry Pi 5 in debug mode...
filesystem_layout.c: icudtl file not found!
Application finished.
```

Is there any way to run in debug mode my application? Looks like native flutter run don't generate engine file icudtl

lexxvis, 2025-01-26:
filesystem_layout.c: icudtl file not found!
I faced with the same issue. Raspberry Pi4. How to solve it?

MaksymTiutiunnyk, 2025-03-19:
Is there still no solution for this issue? I get filesystem_layout.c: icudtl file not found! when trying to run Flutter app on Radxa CM3 with Ubuntu 22.04 LTS. Host machine is Windows 11 with Flutter 3.29 (stable channel). I have the latest available engine binaries on Radxa CM3 as well. Additionally, I have Flutter SDK on the Radxa, but when launching the app using it I experience a poor performance. So, I hoped launching the app with flutter-pi can solve this problem, but it doesn't launch at all)

ardera, 2025-03-19:
> Is there still no solution for this issue? I get filesystem_layout.c: icudtl file not found! when trying to run Flutter app on Radxa CM3 with Ubuntu 22.04 LTS. Host machine is Windows 11 with [...]

Mh, flutter 3.29 is not supported yet, shouldn't even be possible to activate flutterpi_tool with that version. With 3.27 I can't reproduce this

@MaksymTiutiunnyk Can you run `flutterpi_tool run` command with the `-v` flag and send me the output?

MaksymTiutiunnyk, 2025-03-20:
@ardera Since my host machine is Windows I used the second (old) method of running an app, so I did not even try to install flutterpi_tool. I have just tried to install it, but the installation fails.

> With 3.27 I can't reproduce this
Do you mean you have a Radxa CM3 board and you are able to run a Flutter app on it using flutter-pi?

MaksymTiutiunnyk, 2025-03-20:
Well, I have fixed the "filesystem_layout.c: icudtl file not found!" problem by downgrading Flutter to 3.27.0 and completely reinstalling flutter-pi (I don't know for sure what actually helped). But now I get 
[ERROR:flutter/shell/common/shell.cc(115)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(146)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(557)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(668)] Could not launch engine with configuration.

I saw the same [here](https://github.com/ardera/flutter-pi/issues/179). It seems that engine binaries are not up-to-date

ardera, 2025-03-21:
> Do you mean you have a Radxa CM3 board and you are able to run a Flutter app on it using flutter-pi?

No, but I have a bunch of other embedded boards here. Though that shouldn't even matter, if it's an embedded linux board, it's all the same as far as flutterpi_tool is concerned.

> [@ardera](https://github.com/ardera) Since my host machine is Windows I used the second (old) method of running an app, so I did not even try to install flutterpi_tool. I have just tried to install it, but the installation fails.

Well, I just released the update for flutter 3.29 compatibility yesterday, so maybe you tried before I released it. I just tried on Windows and it works fine for me. I did stumble upon some issues with executable permissions once the app was installed on target though, which I just fixed in 0.7.1. So Windows definitely should work fine.

> Well, I have fixed the "filesystem_layout.c: icudtl file not found!" problem by downgrading Flutter to 3.27.0 and completely reinstalling flutter-pi (I don't know for sure what actually helped). But now I get
[ERROR:flutter/shell/common/shell.cc(115)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(146)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(557)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(668)] Could not launch engine with configuration.

You need to use the correct binaries for the SDK. If you're using Flutter 3.27.0, you need to use the Flutter 3.27.0 engine binaries, which are not the ones that will be installed if you just clone the engine-binaries repo as-is. For 3.27.0, the correct binaries are: https://github.com/ardera/flutter-engine-binaries-for-arm/tree/engine_83bacfc52569459a4a654727cad2546820cb0d6a.

Though, in general, these engine binaries are mostly provided just for backwards compatibility. Nowadays flutterpi_tool should be the better way to build & run your app. In fact, the exact reason I wrote it is because a lot of people had the same version mismatch problems that you had when installing the engine binaries manually. Maybe you can give flutterpi_tool another try, now that it's compatible to 3.29 (and works better on windows), and let me know if something's not working. :)

MaksymTiutiunnyk, 2025-03-21:
Thank you for such a great answear! I did not use flutterpi_tool because I was confused by this line in the README.md:
_Building the App (New Method, Linux-only)_

I thought it was about host machine, and not about target platform) When I tried to use flutterpi_tool it worked just fine for me, and the engine binaries problem disappeared. So, I would suggest you to change this description a little bit, cause I think other people may also be confused and not use flutterpi_tool for that reason.

The only problem I have now is the one mentioned [here](https://github.com/philseeley/boatinstrument/issues/3) and in a few other issues. That is what I get when running my app:
rock@radxa-cm3-rpi-cm4-io:~$ sudo flutter-pi test/flutter_assets/
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
arm_release_ver: g24p0-00eac0, rk_so_ver: 7
flutter: The Dart VM service is listening on http://127.0.0.1:34473/S2t0wZuwf4Q=/

And that is what I get when moving mouse:
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88

I had a disabled mouse cursor on the board, but I have already enabled it and the problem persists