ardera/flutter-pi #379: Having Issues with flame_audio plugin. What am I missing?

MatthewJones517, 2024-01-15:
I am trying to get flutter-pi working with something I whipped with the flame and flame_audio packages. I've got it 95% of the way there. I get the following error every time a sound plays:

```
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method create on channel xyz.luan/audioplayers)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:320)
<asynchronous suspension>
#1      AudioplayersPlatform.create (package:audioplayers_platform_interface/src/audioplayers_platform.dart:18)
<asynchronous suspension>
#2      AudioPlayer._create (package:audioplayers/src/audioplayer.dart:161)
<asynchronous suspension>
```

I found the following issue that appears to be related:
https://github.com/ardera/flutter-pi/issues/120

It seems like the flame_audio package uses audioplayer as a dependency. Given I've seen several threads about it here, I'm guessing it's related to flutter_pi. Especially everything works fine on my Ubuntu dev machine. Do you have any suggestions as to what the issue may be?

MatthewJones517, 2024-01-15:
Okay, I'm 99% sure I've sorta found the root cause on this.

It looks like pull request [345](https://github.com/ardera/flutter-pi/pull/345) is still open and it implies that audioplayers v5 isn't supported in the current release. 

Looking at the flame_audio package, it [appears to use](https://github.com/flame-engine/flame/blob/622c855318b6c1731891b023ddc6429ba1f32329/packages/flame_audio/pubspec.yaml#L16C3-L16C23) audioplayers 5. 

Any idea on when that PR is going to get merged in and released? I'm trying to decide what direction to take my project. :)

ardera, 2024-01-15:
Hey, you're right, #345 was the cause of this. Just merged it 😄

MatthewJones517, 2024-01-15:
Thanks so much! I appreciate it. :)

closed 2024-01-15 by MatthewJones517