ardera/flutter-pi #380: Audio Devices Quickly Being Exhausted

MatthewJones517, 2024-01-16:
This is a follow-up to my previous issue #379. After the merging of PR #345 the audio does indeed work. However I'm getting a new issue. 

I'm using Flame, and triggering a bunch of audio sfx in short succession. The first few work fine, then it goes silent and I get the following error:

`AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': Device or resource busy`

Initially I thought Flame just wasn't releasing the device after playing the sfx, but it works fine on a variety of other devices. This includes my Ubuntu development box. 

This makes me think that flutter-pi isn't properly releasing the audio resources when asked. I can provide a code sample if necessary.

ardera, 2024-01-16:
Ah I see why, audioplayers `ReleaseMode.release` is not implemented in flutter-pi. So the native audio player is not disposed on end of playback, as it should. Though should not be hard to implement

MatthewJones517, 2024-01-16:
Thanks so much for looking at this! I was going to do more debugging tonight after work, but it appears you already found the root cause. :)

ardera, 2024-01-16:
Okay, seems like iOS doesn't implement it either (https://github.com/bluefireteam/audioplayers/issues/56), still would be good to have in flutter-pi. But maybe you can work around for now using AudioPool: https://pub.dev/documentation/audioplayers/latest/audioplayers/AudioPool-class.html

MatthewJones517, 2024-01-16:
AudioPool was the next thing I was going to try. I suspect it will be successful, and is probably the right call for my use case regardless. :)

MatthewJones517, 2024-01-21:
AudioPool does allow the clips to be played. That being said, any audio trigger generates the following warning:

`plugins/audioplayers/player.c: Could not query current position.`

The audio still plays correctly, but the warning doesn't appear on my Ubuntu devbox.

ardera, 2024-01-22:
Interesting, in your test app, do you have a way to check the current playback position reported by the audio player? Can you verify if it matches the actual playback position of the audio?

MatthewJones517, 2024-01-22:
Unfortunately not. They're just sound effects being kicked off by my "game". (It uses flame and flame_audio, but calling this thing a game is too generous 😂)

In researching the error, it appears to happen when audio_players are used outside of a context. So is it possible that flutter-pi isn't properly providing context to audio_players?

DoumanAsh, 2024-02-01:
> plugins/audioplayers/player.c: Could not query current position.

This is often the case for short sound effecs because you effectively cannot get correction position when your overall duration is within second or two.
I wouldn't consider it critical issue, but more to do with callback nature of player itself
This is likely happens at the end of audio

> This makes me think that flutter-pi isn't properly releasing the audio resources when asked. I can provide a code sample if necessary.

I'm not sure how `flame` works, but one way to clean up resources is to delete player via manual [dispose](https://pub.dev/documentation/audioplayers/latest/audioplayers/AudioPlayer/dispose.html) call

To release resources, `audioplayers` make `release` call
https://github.com/ardera/flutter-pi/blob/master/src/plugins/audioplayers/plugin.c#L78C9-L78C29

But this is to clean up whole player

The reason you get exhaust error is probably of incorrect usage of `AudioPlayer` within that `flame` thingy

I remember encountering this issue by constantly creating new player and playing sound one time.
You obviously will reach issue because VM nature does not allow to quickly dispose of temporary player.

So the only proper solution is to make sure you either re-use player or call `AudioPlayer::dispose` every time you no longer need instance

MatthewJones517, 2025-12-02:
I just realized I never closed this issue. 

For anyone who finds this in the future, this is an issue with the Pi itself, not `flutter-pi`. Check out `flutter_soloud` as a potential solution.

closed 2025-12-02 by MatthewJones517