ardera/flutter-pi #382: video_player on raspberry pi 4 64 bit

beniskopf, 2024-01-20:
Hi, i'm trying to play video files on a raspberry pi 4 (64 bit). As stated in the docs i should use 
`$ sudo apt install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-plugins-bad gstreamer1.0-libav gstreamer1.0-alsa` + [https://pub.dev/packages/video_player](url). however it gives me this error message. any ideas how to fix this? maybe related to [https://github.com/ardera/flutter-pi/issues/360](url)

> [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: UnimplementedError: init() has not been implemented.
#0      VideoPlayerPlatform.init (package:video_player_platform_interface/video_player_platform_interface.dart:43)
#1      _videoPlayerPlatform (package:video_player/video_player.dart:28)
#2      VideoPlayerController.initialize (package:video_player/video_player.dart:430)
#3      _BumbleBeeRemoteVideoState.initState (package:miata_screen/filepicker.dart:117)
#4      StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5602)
#5      ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
#6      Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#7      Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#8      SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6736)
#9      Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#10     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#11     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#12     Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#13     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
#14     ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
#15     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#16     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#17     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6736)
#18     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#19     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#20     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6736)
#21     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#22     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#23     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#24     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
#25     Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#26     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
#27     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5625)
#28     ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
#29     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#30     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#31     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6736)
#32     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#33     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#34     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#35     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
#36     Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#37     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
#38     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5625)
#39     ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
#40     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#41     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#42     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6736)
#43     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#44     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#45     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#46     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
#47     Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#48     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
#49     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5625)
#50     ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
#51     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#52     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#53     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#54     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
#55     Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#56     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
#57     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5625)
#58     ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
#59     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#60     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#61     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6736)
#62     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
#63     Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
#64     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#65     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
[... 198 lines cut ...]
#264    Element.updateChild (package:flutter/src/widgets/framework.dart:3815)
#265    SingleChildRenderObjectElement.update (package:flutter/src/widgets/framework.dart:6743)
#266    Element.updateChild (package:flutter/src/widgets/framework.dart:3815)
#267    SingleChildRenderObjectElement.update (package:flutter/src/widgets/framework.dart:6743)
#268    Element.updateChild (package:flutter/src/widgets/framework.dart:3815)
#269    ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#270    Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#271    ProxyElement.update (package:flutter/src/widgets/framework.dart:5800)
#272    Element.updateChild (package:flutter/src/widgets/framework.dart:3815)
#273    ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#274    Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#275    ProxyElement.update (package:flutter/src/widgets/framework.dart:5800)
#276    Element.updateChild (package:flutter/src/widgets/framework.dart:3815)
#277    ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5496)
#278    StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
#279    Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
#280    BuildOwner.buildScope (package:flutter/src/widgets/framework.dart:2895)
#281    WidgetsBinding.drawFrame (package:flutter/src/widgets/binding.dart:984)
#282    RendererBinding._handlePersistentFrameCallback (package:flutter/src/rendering/binding.dart:457)
#283    SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1325)
#284    SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1255)
#285    SchedulerBinding._handleDrawFrame (package:flutter/src/scheduler/binding.dart:1113)
#286    _invoke (dart:ui/hooks.dart:312)
#287    PlatformDispatcher._drawFrame (dart:ui/platform_dispatcher.dart:383)
#288    _drawFrame (dart:ui/hooks.dart:283)

ardera, 2024-01-22:
You need to depend on the `flutterpi_gstreamer_video_player` package and call `FlutterpiVideoPlayer.registerWith()` in the main, before the `runApp` call: https://pub.dev/packages/flutterpi_gstreamer_video_player

I'll fix the readme instructions to mention that.

beniskopf, 2024-01-22:
thanks for your help. I've included the `flutterpi_gstreamer_video_player` package and called `FlutterpiVideoPlayer.registerWith()`.  now i get this error message: 

```
> rp@raspberrypi:~ $ sudo flutter-pi --release -r 270 my_apps_flutter_assets/
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
plugins/gstreamer_video_player/plugin.c: Couldn't create gstreamer video player.
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(nativeerror, Input/output error, 5, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:322)
<asynchronous suspension>
#2      FlutterpiVideoPlayer.create (package:flutterpi_gstreamer_video_player/src/platform.dart:78)
<asynchronous suspension>
#3      VideoPlayerController.initialize (package:video_player/video_player.dart:434)
<asynchronous suspension>
#4      _VideoPlayerScreenState.initState.<anonymous closure> (package:miata_screen/filepicker.dart:104)
<asynchronous suspension>
```

i tried to play the video using `gst-launch-1.0 playbin uri=file:///home/rp/media/mtv.mp4` and it works but the colors are distorted.

ardera, 2024-01-23:
Are you on the latest revision of flutter-pi? This looks like a bug I fixed 3 weeks ago :)

beniskopf, 2024-01-25:
sorry i'm retarded. `_controller.play();` was missing.

closed 2024-01-25 by beniskopf