ardera/flutter-pi #383: pi@raspberrypi:~ $ flutter-pi ./test/ filesystem_layout.c: icudtl file not found!

gnu-xiaosong, 2024-01-27:
pi@raspberrypi:~ $ flutter-pi ./test/
filesystem_layout.c: icudtl file not found!

why？

SwordTechCorp, 2024-01-28:
You need to install engine binary. You can either build it yourself or just follow the instruction provided by the author
[repo](https://github.com/ardera/flutter-engine-binaries-for-arm)

It's mentioned in the readme btw.

manszigher, 2024-04-19:
I am faced with the same problem I am building using the yocto project and have bumped up flutter to 3.19.3. When installing flutter 3.19.3 we are getting the following files installed

```
flutter-engine:
       /usr/share/flutter/3.19.3/debug/data/icudtl.dat
       /usr/share/flutter/3.19.3/debug/lib/libflutter_engine.so
       /usr/share/flutter/3.19.3/profile/data/icudtl.dat
       /usr/share/flutter/3.19.3/profile/lib/libflutter_engine.so
       /usr/share/flutter/3.19.3/release/data/icudtl.dat
       /usr/share/flutter/3.19.3/release/lib/libflutter_engine.so
```

From looking at the flutter-pi code

https://github.com/ardera/flutter-pi/blob/master/src/filesystem_layout.c#L97

it looks like it will only look for the engine directly under /usr/share/flutter so it cannot locate the icudtl file.  Now I can change it so that we skip the version and only install under /usr/share/flutter but flutter-pi will still not find the file because in this version it is placed under the data folder which was not the case earlier. So is my installation of flutter-engine broken or can I tell flutter-pi where to look for the icudtl file?

manszigher, 2024-04-19:
I am new to the flutter eco-system but I am using this https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-engine/flutter-engine_git.bb to install the flutter-engine.

ardera, 2024-04-22:
@manszigher It will look in `/usr/share/flutter`, but also in `data/icudtl.dat` relative to the bundle path, at least when the meta-flutter filesystem layout is chosen at build-time. (Which meta-flutter will do for you implicitly)

meta-flutter will add symlinks at those locations: https://github.com/meta-flutter/meta-flutter/blob/kirkstone/conf/include/flutter-app.inc#L54-L59

What commandline are you launching flutter-pi with?

officialluq, 2024-08-29:
I think I know what may cause that error. Did you use meta-flutter? (yocto)