ardera/flutter-pi #384: "could not commit frame request"

erikms61, 2024-01-30:
Hi Ardera,
First of all, you have done an amazing job at making Flutter available on the Raspberry Pi.
Platform is Raspberry Pi 2 Model B Rev 1.1, with 2 GB of RAM
`uname -a` output is
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
However... I run into the error messages
`modesetting.c: commit requested, bt drmdev is paused right now`
followed by
`window.c: Could not commit frame request`

Building on debian PC,
`flutterpi_tool build --release` produces the `build/flutter_assets` folder, no errors
copy it over to the RPi
try to run it with `flutter-pi --release ./flutter_assets`
and run into the above-mentioned errors.
Is the Pi2 simply not capable of running this, or am I missing something?

ardera, 2024-02-06:
Hey, yes you need to boot into console: https://github.com/ardera/flutter-pi#configuring-your-raspberry-pi

Only one application can grab the display output at a time. So if a desktop environment is running, flutter-pi can't grab it.

galets, 2024-09-01:
is it possible to crash when such condition occurs? I understand that a console is required, but the issue is: process reports an error, but never crashes. If it exited with an error, that would be a correct behavior, because then it could be restarted. But (at least in my case) the app never crashes, just stays running without showing anything.