ardera/flutter-pi #385: Gstreamer playing without sound

Binozo, 2024-02-07:
Hello there 👋

lately I made some experiments with the newest version of flutter-pi and the `Gstreamer` video player. I tried playing a video which succeeded, but sadly I got no sound out of it. I verified that the video actually contains sound (played it on my pc) but neither my Raspberry Pi 4 nor Raspberry Pi zero 2 w are able to put audio through my TV. (And yes I verified that my TV is not in mute mode)

Do you know any solutions to this?

ardivee, 2024-03-18:
I have the same issue here, the audioplayers package works fine if i try to play sounds ( Wanted to make sure i could actually get some sound ).

I also have tried running gstreamer on the desktop ( Raspberry Pi ), and there i can also just play the video fine with sound.

edlman, 2025-01-27:
I have the same issue on Flutter Pi on RPi 5 running Debian 12 (Bookworm) and RPi4 running Debian 11 (Bullseye). Playing the same video file with `gst123` (gstreamer player) is fine (video and audio).

ardera, 2025-11-16:
Hey, sorry for not replying to this. This was actually just a missing feature, it's implemented in the devel branch of flutter-pi in combination with the corresponding branch of the flutter_packages repo:
- https://github.com/ardera/flutter-pi/tree/devel
- https://github.com/ardera/flutter_packages/tree/feat/video-player-audio