ardera/flutter-pi #386: Running on Banana Pi M2+

dpmilian, 2024-02-10:
I've been using flutter-pi on a banana pi m2+ running Armbian 22.11.1 Jammy succesfully for a while, using the flutter-pi release from January (commit 20483a1195c88bd44ff48be45582deb8cc3b1fb7). After upgrading the dart and flutter sdks, I also went along and upgraded flutter-pi to the latest version (as of writing 75e72fef8e8ce65ec72a002c00d9c6db4d1479d3). 

Unfortunately, it is no longer rendering, I still get a running app (backend services are active and connecting to mqtt broker for example) but no rendered output on screen.

Building 'flutter-pi itself is run on the m2+ following the same instructions as before: https://github.com/ardera/flutter-pi?tab=readme-ov-file#-building-flutter-pi-on-the-raspberry-pi

My Makefile targets for building the flutter app are:
```
build:
	flutter build bundle

build-release-stage-dart:
	dart ${DART_SDK_FOLDER}/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot --sdk-root ${DART_SDK_FOLDER}/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ --target=flutter --aot --tfa -Ddart.vm.product=true --packages .dart_tool/package_config.json --output-dill build/kernel_snapshot.dill --verbose --depfile build/kernel_snapshot.d package:naust_hmi/main.dart

build-release: build build-release-stage-dart
	./engine-binaries/arm/gen_snapshot_linux_x64_release --deterministic --snapshot_kind=app-aot-elf --elf=build/flutter_assets/app.so --strip --sim-use-hardfp build/kernel_snapshot.dill

build-new: build
	flutter pub global run flutterpi_tool build --release --arch=arm
```
So I can choose the old and new building method.

**Before**  the update, running the app (screen rendered correctly) showed the following logs:
``` console
/usr/local/bin/flutter-pi --release -o portrait_up /home/scr/Projects/flutter_assets/
[modesetting] Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
[flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
==============Locale==============
Flutter locale:
  default: es_ES
  locales: es_ES es.UTF-8 es.UTF-8 es.UTF-8 es_ES es es.UTF-8
===================================
===================================
display mode:
  resolution: 960 x 544
  refresh rate: 61Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1,000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_EXT_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_partial_update EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 22.2.5"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "lima"
  renderer: "Mali400"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_EXT_blend_func_extended GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_EXT_multisampled_render_to_texture GL_EXT_multisampled_render_to_texture2 GL_EXT_clip_control GL_KHR_parallel_shader_compile GL_MESA_bgra "
===================================
```

**After** the update, the logs look like:
``` console
 /usr/local/bin/flutter-pi --release -o portrait_up /home/scr/Projects/flutter_assets/
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
(...)
```

The underlying hw, OS and installed libraries has not changed, in theory the only updates have to do with flutter/ dart, the flutter-engine-binaries-for-arm and the flutter-pi binaries themselves.

Any suggestions of where I could start looking at debugging or fixing this issue? The error message points at https://github.com/ardera/flutter-pi/blob/75e72fef8e8ce65ec72a002c00d9c6db4d1479d3/src/modesetting.c#L2323

ardera, 2024-02-12:
Interesting, I actually have a Banana Pi M2 Zero here (also running Armbian Jammy) and it seems to work fine here.

Can you build flutter-pi in debug mode & with drm debugging enabled, run again, and paste the complete cmdline output? (Configure with `-DCMAKE_BUILD_TYPE=Debug -DDEBUG_DRM_PLANE_ALLOCATIONS=ON`) 

Also a `drminfo` snapshot would be great: https://drmdb.emersion.fr/

Thanks!

dpmilian, 2024-02-14:
Run on commit 75e72fef8e8ce65ec72a002c00d9c6db4d1479d3:

`cmake -DCMAKE_BUILD_TYPE=Debug -DDEBUG_DRM_PLANE_ALLOCATIONS=ON ..`

``` bash
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Debug
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libdrm'
--   Found libdrm, version 2.4.113
-- Checking for module 'gbm'
--   Found gbm, version 23.0.4-0ubuntu1~22.04.1
-- Checking for module 'libsystemd'
--   Found libsystemd, version 249
-- Checking for module 'libinput'
--   Found libinput, version 1.20.0
-- Checking for module 'xkbcommon'
--   Found xkbcommon, version 1.4.0
-- Checking for module 'libudev'
--   Found libudev, version 249
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'egl'
--   Found egl, version 1.5
-- Checking for module 'glesv2'
--   Found glesv2, version 3.2
-- EGL/GLES support ....... ON
-- Lint EGL headers ....... OFF
-- Checking for module 'vulkan'
--   No package 'vulkan' found
-- Vulkan support ......... OFF
-- Filesystem Layout ...... default
-- Checking for module 'libseat'
--   No package 'libseat' found
libseat was not found. flutter-pi will be built without session switching support.
-- Session switching ...... OFF
-- Checking for module 'gstreamer-1.0'
--   Found gstreamer-1.0, version 1.20.3
-- Checking for module 'gstreamer-plugins-base-1.0'
--   Found gstreamer-plugins-base-1.0, version 1.20.1
-- Checking for module 'gstreamer-app-1.0'
--   Found gstreamer-app-1.0, version 1.20.1
-- Checking for module 'gstreamer-allocators-1.0'
--   Found gstreamer-allocators-1.0, version 1.20.1
-- Checking for module 'gstreamer-video-1.0'
--   Found gstreamer-video-1.0, version 1.20.1
-- Checking for module 'gstreamer-audio-1.0'
--   Found gstreamer-audio-1.0, version 1.20.1
-- Performing Test COMPILER_SUPPORTS_MACRO_PREFIX_MAP
-- Performing Test COMPILER_SUPPORTS_MACRO_PREFIX_MAP - Success
-- IPO/LTO ................ OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /home/scr/debug/flutter-pi/build
```
-----------------------------
And then, output:
```
./flutter-pi --release -o portrait_up /home/scr/Projects/flutter_assets
filesystem_layout.c: icudtl file not found at /home/scr/Projects/flutter_assets/icudtl.dat.
filesystem_layout.c: icudtl file not found at /usr/share/flutter/icudtl.dat.
[... 35 lines cut ...]
egl_gbm_render_surface.c: using fourcc AR24 (ARGB 8:8:8:8) with modifier 0x0
modesetting.c:   checking if plane with id 31 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false
modesetting.c:   checking if plane with id 37 qualifies...
modesetting.c:     does not qualify: plane does not support the modified format XRGB 8:8:8:8, 0.
modesetting.c:   checking if plane with id 41 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false
modesetting.c:   checking if plane with id 45 qualifies...
modesetting.c:     does not qualify: plane type is cursor but allow_cursor is false
modesetting.c: Could not find a suitable unused DRM plane for pushing the framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
modesetting.c:   checking if plane with id 31 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false
modesetting.c:   checking if plane with id 37 qualifies...
modesetting.c:     does not qualify: plane does not support the modified format XRGB 8:8:8:8, 0.
modesetting.c:   checking if plane with id 41 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false
modesetting.c:   checking if plane with id 45 qualifies...
modesetting.c:     does not qualify: plane type is cursor but allow_cursor is false
```
Find attached the output of drminfo:

[drm_info_armbian_bpi_m2plus.json](https://github.com/ardera/flutter-pi/files/14277693/drm_info_armbian_bpi_m2plus.json)

Thanks!

ardera, 2024-02-14:
Seems like your display controller driver is broken somehow, it reports an empty list of supported formats:
```js
// ... 
"IN_FORMATS": {
  "id": 30,
  "flags": 20,
  "type": 16,
  "atomic": false,
  "immutable": true,
  "raw_value": 42,
  "spec": null,
  "value": null,
  "data": [
    {
      "modifier": 0,
      "formats": [
      ]
    }
  ]
},
```

Though, in some other place it correctly reports the supported formats. It's just that the IN_FORMATS property is the more accurate way of checking normally so flutter-pi uses that

Can you maybe try with a newer kernel? If not, I can add a workaround in flutter-pi.

dpmilian, 2024-02-14:
My current kernel version is: 5.15.80-sunxi #22.11.1

After upgrading to 6.1.63-current-sunxi, the result is similar:
``` bash
./flutter-pi --release -o portrait_up /home/scr/Projects/flutter_assets
filesystem_layout.c: icudtl file not found at /home/scr/Projects/flutter_assets/icudtl.dat.
filesystem_layout.c: icudtl file not found at /usr/share/flutter/icudtl.dat.
==============Locale==============
Flutter locale:
  default: es
  locales: es es.UTF-8 es_ES es.UTF-8 es.UTF-8 es_ES es.UTF-8
===================================
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
===================================
EGL information:
  version: 1.4
  vendor: Mesa Project
  client extensions: EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_EXT_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless
  display extensions: EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_partial_update EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display 
===================================
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 23.0.4-0ubuntu1~22.04.1"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Mesa"
  renderer: "Mali400"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_debug_label GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_EXT_blend_func_extended GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_EXT_multisampled_render_to_texture GL_EXT_multisampled_render_to_texture2 GL_EXT_texture_compression_s3tc_srgb GL_EXT_clip_control GL_KHR_parallel_shader_compile GL_MESA_bgra "
===================================
window.c: WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display. 
Use the `-d` commandline parameter to specify the physical dimensions of your display.
display mode:
  resolution: 1080 x 1920
  refresh rate: 59.194215Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
  pixel format: (any)
pluginregistry.c: Initialized plugins: services, text input, raw keyboard plugin, gstreamer video_player, audioplayers, 
window.c: Flutter requested render surface before supplying surface dimensions.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
gl_renderer.c: Choosing EGL config with pixel format ARGB 8:8:8:8...
egl_gbm_render_surface.c: using fourcc AR24 (ARGB 8:8:8:8) with modifier 0x0
modesetting.c:   checking if plane with id 31 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false
modesetting.c:   checking if plane with id 37 qualifies...
modesetting.c:     does qualify.
modesetting.c:   checking if plane with id 31 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false
modesetting.c:   checking if plane with id 37 qualifies...
modesetting.c:     does qualify.
modesetting.c:   checking if plane with id 31 qualifies...
modesetting.c:     does not qualify: plane type is overlay but allow_overlay is false

```

dpmilian, 2024-03-12:
Yes, you are correct (sorry for the three weeks delay). I was distracted by the amount of debug messages and I didn't have the screen in front of me to verify it was effectively working.

For reference, with the installed package versions mentioned in the cmake output [from my previous comment](https://github.com/ardera/flutter-pi/issues/386#issuecomment-1943388263) and with kernel version 6.1.63-current-sunxi,  it works as expected.

Thanks for your help!

closed 2024-03-12 by dpmilian