ardera/flutter-pi #387: Feature request: bash script installer

payam-zahedi, 2024-02-20:
The installation process could be tricky for the first time. So I wanted to suggest adding a bash script install file to use install all the requirments.

payam-zahedi, 2024-04-18:
@ardera Whats your opinion on this?

Do you think it makes sense? Are there any blockers or challenges here?

I can work on it

ardera, 2024-04-18:
@payam-zahedi Ah, sry. I think in the ideal case, people shouldn't even have to build flutter-pi themselves. It could just be built by CI and then bundled with the app by flutterpi_tool.

payam-zahedi, 2024-04-18:
We made snapp_cli, its a way that you can install flutter on your remote device and create a custom device and connect it to flutter sdk. 

For now its only works flutter 

But, I have a plan to integrate flutter-pi also to it

payam-zahedi, 2024-04-18:
Do you think 

Can we make a reusable bash script that can be independent of hardware?