ardera/flutter-pi #389: Failed to build flutterpi_tool

MihirGrand, 2024-02-27:
I ran the command `flutter pub global activate flutterpi_tool` but I get the following error

Failed to build flutterpi_tool:flutterpi_tool:
.pub-cache/hosted/pub.dev/flutterpi_tool-0.3.0/lib/src/cache.dart:297:20: Error: The type 'HostPlatform' is not exhaustively matched by the switch cases since it doesn't match 'HostPlatform.windows_arm64'.
 - 'HostPlatform' is from 'package:flutter_tools/src/base/os.dart' ('flutter/packages/flutter_tools/lib/src/base/os.dart').
Try adding a wildcard pattern or cases that match 'HostPlatform.windows_arm64'.
    return switch (host) {

Trying this on Ubuntu through WSL

ardera, 2024-02-27:
flutterpi_tool is only supported for flutter stable right now, can you make sure you're on stable?

MihirGrand, 2024-02-27:
I wasn't on stable, I was on main 😅
It works now. Thanks!

closed 2024-02-27 by MihirGrand