ardera/flutter-pi #394: No issue, just looking for some help: Can't load Kernel binary: Invalid kernel binary format version.

Osmosis311, 2024-03-06:
Hi!  I love this library.  It's fantastic.  Thank you so much.  

I'm having some trouble getting it working on my Pi.  I'm sure it's one obvious thing I'm missing, but I've tried for a few hours and can't seem to figure it out.

1. I am on a Mac M1
2. Below is the output from Flutter doctor on my mac
3. On my Mac, I am running this command, and it completes successfully: `flutterpi_tool build --arch=arm64 --cpu=pi4`
4. On my Pi, I downloaded the latest version of flutter-pi, did make and install.
5. I copied the flutter_assets folder to the Pi, and when I run `flutter-pi flutter-assets/`, I get the error below.

Any advice would be _greatly_ appreciated!  Thank you so much!

Flutter Doctor output (on my Mac):
`Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.19.2, on macOS 14.3 23D56 darwin-arm64, locale en-US)
[✓] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 15.2)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2022.3)
[✓] VS Code (version 1.87.0)
[✓] Connected device (6 available)
[✓] Network resources`

Error when running `flutter-pi flutter-assets/`:

`xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:39:34: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:40:29: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:41:29: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:42:29: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:43:29: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:44:27: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:45:27: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:46:27: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:47:27: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:48:29: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:49:29: string literal is not a valid UTF-8 string
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:49:29: too many errors
xkbcommon: ERROR: /usr/share/X11/locale/iso8859-1/Compose:49:29: failed to parse file
keyboard.c: Could not create compose table from locale.
user_input.c: Could not initialize keyboard configuration. Flutter-pi will run without text/raw keyboard input.
[ERROR:flutter/shell/common/shell.cc(91)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
[ERROR:flutter/runtime/dart_isolate.cc(144)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(422)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(614)] Could not launch engine with configuration.
flutter: The Dart VM service is listening on http://127.0.0.1:39355/2VZdM3cuXBY=/`

Osmosis311, 2024-05-01:
Anyone encounter the same issue?

MathavK-raptee, 2024-09-12:
Is file in elf format?
Try file libapp.so or app.so

vanlooverenkoen, 2024-11-06:
I have the same issue