ardera/flutter-pi #404: 🐛 throwing DRM and GBM errors with newest kernel

lukas-runge, 2024-04-13:
Hi folks,

I just updated my Pi 5 to the latest kernel version with `sudo apt update && sudo apt upgrade`.
I am now running on: `6.6.26-v8-16k+`

Sadly flutter-pi just crashes now - every time I try to start it in release mode.

```
> flutter-pi --release build/flutter_assets

modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
flutter: {}
modesetting.c: Couldn't add buffer as DRM fb. drmModeAddFB2WithModifiers: Invalid argument
egl_gbm_render_surface.c: Couldn't add GBM buffer as DRM framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
```

Any ideas on how to solve that?

Kind regards,
@lukas-runge

lukas-runge, 2024-04-13:
Just rolled back the Raspberry Pi 5 to [Kernel v6.1.77](https://github.com/Hexxeh/rpi-firmware/commit/5fc4f643d2e9c5aa972828705a902d184527ae3f) via `sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f`. 🚧

After a reboot flutter-pi works perfectly again. 🎉
Non the less a fix would be great to support newer kernel versions in the future. 🙌

**Best regards,
@lukas-runge**

DonatelloX, 2024-05-01:
I have the same problem.
Please, can you fix it?

ardera, 2024-05-01:
@lukas-runge @DonatelloX What display are you using and how is it connected? If you have a DSI display, can you try with latest master? I just merged a fix for a Pi 5 DSI problem yesterday: #416 .

DonatelloX, 2024-05-02:
The problem seems solved now.

DonatelloX, 2024-05-03:
It does not always appear but sometimes immediately after launching the application:
```
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card0". Continuing.
```
Last kernel 6.6.28

kyleyannelli, 2024-12-21:
I'm getting

```
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
```

as well on my Pi 4B, Debian 12, Kernel 6.6.62-v8+

ardera, 2025-01-13:
> On my Pi 4B, Debian 12, Kernel 6.6.62-v8+, in the console I'm getting
> 
> ```
> modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
> flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
> ```
> 
> Here are my groups
> 
> `adm dialout cdrom sudo audio video plugdev games users input render netdev lpadmin gpio i2c spi`
> 
> This is out from HDMI 1 port to a 1080p an old Samsung TV. I tried both KMS and FKMS.

That's just normal output (but I agree it is a bit confusing, it looks like an error). I'll remove that in a future version