ardera/flutter-pi #408: MissingPluginException - method setSourceBytes (audiplayers)

bbence84, 2024-04-15:
I'm using audioplayers 5.0.0 in my Flutter project, and I'm getting MissingPluginException for the setSourceBytes method.
According to the feature parity table here: https://github.com/bluefireteam/audioplayers/blob/main/feature_parity_table.md, this way of loading a resource should also work on linux (byte array).

Any ideas? 

Thanks!

bbence84, 2024-04-15:
Maybe this is related?
https://github.com/bluefireteam/audioplayers/issues/1154
Sorry, I am not really familiar with flutter-pi yet, and how it builds the projects to rpi...

ardera, 2024-04-16:
Problem is that it's missing an implementation for `setSourceBytes` here: https://github.com/ardera/flutter-pi/blob/master/src/plugins/audioplayers/plugin.c

"linux" as a flutter plugin platform means "linux desktop", e.g. the glfw embedder, not flutter-pi, ivi-homescreen, sony embedder, etc. Custom embedders have to build the native code for any plugins they wish to support themselves