ardera/flutter-pi #412: No output on Odroid M1S when using Vu5S MIPI-DSI display

TeunVR, 2024-04-23:
Running flutter-pi on an Odroid M1S board (Rockchip RK3566). 
It runs really great and HDMI output is working fine.
However i cannot get the Vu5S MIPI-DSI display to work (https://wiki.odroid.com/accessory/display/vu_series/vu5s).
The application starts ok and it detects the display correctly. Everything seems to be ok, it is just not rendering anything. The display keeps showing console output. 
I tried [kmscube](https://gitlab.freedesktop.org/mesa/kmscube) from the meta project and that is working fine. Also gnome is working. 

Running on Odroid Ubuntu 20.04 image.
 
```
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
==============Locale==============
Flutter locale:
  default: C
  locales: C C.UTF-8 C.UTF-8 C.UTF-8 C.UTF-8
===================================
device[0]
	available_nodes 0005
	nodes
		nodes[0] /dev/dri/card0
		nodes[2] /dev/dri/renderD128
	bustype 0002
	businfo
Unknown/unhandled bustype: 2

flutter-pi.c: New dev from path: /dev/dri/card0
modesetting.c: Unknown DRM connector property: brightness
modesetting.c: Unknown DRM connector property: contrast
modesetting.c: Unknown DRM connector property: saturation
modesetting.c: Unknown DRM connector property: hue
modesetting.c: Unknown DRM crtc property: SOC_ID
modesetting.c: Unknown DRM crtc property: PORT_ID
modesetting.c: Unknown DRM crtc property: ACLK
modesetting.c: Unknown DRM crtc property: BACKGROUND
modesetting.c: Unknown DRM crtc property: LINE_FLAG1
modesetting.c: Unknown DRM crtc property: left margin
modesetting.c: Unknown DRM crtc property: right margin
modesetting.c: Unknown DRM crtc property: top margin
modesetting.c: Unknown DRM crtc property: bottom margin
modesetting.c: Unknown DRM crtc property: PLANE_MASK
modesetting.c: Unknown DRM crtc property: FEATURE
modesetting.c: Unknown DRM crtc property: OUTPUT_WIDTH
modesetting.c: Unknown DRM crtc property: OUTPUT_DCLK
modesetting.c: Unknown DRM crtc property: variable refresh rate
modesetting.c: Unknown DRM crtc property: max refresh rate
modesetting.c: Unknown DRM crtc property: min refresh rate
modesetting.c: Unknown DRM crtc property: CUBIC_LUT
modesetting.c: Unknown DRM crtc property: CUBIC_LUT_SIZE
modesetting.c: Unknown DRM crtc property: SOC_ID
modesetting.c: Unknown DRM crtc property: PORT_ID
modesetting.c: Unknown DRM crtc property: ACLK
modesetting.c: Unknown DRM crtc property: BACKGROUND
modesetting.c: Unknown DRM crtc property: LINE_FLAG1
modesetting.c: Unknown DRM crtc property: left margin
modesetting.c: Unknown DRM crtc property: right margin
modesetting.c: Unknown DRM crtc property: top margin
modesetting.c: Unknown DRM crtc property: bottom margin
modesetting.c: Unknown DRM crtc property: PLANE_MASK
modesetting.c: Unknown DRM crtc property: FEATURE
modesetting.c: Unknown DRM crtc property: OUTPUT_WIDTH
modesetting.c: Unknown DRM crtc property: OUTPUT_DCLK
modesetting.c: Unknown DRM crtc property: variable refresh rate
modesetting.c: Unknown DRM crtc property: max refresh rate
modesetting.c: Unknown DRM crtc property: min refresh rate
flutter-pi.c: Found connector: 164 16
flutter-pi.c: gbm_device: armsoc
flutter-pi.c: Renderer type: 0
===================================
EGL information:
  version: 1.4 Bifrost-"r25p0-01eac0"
  vendor: ARM
  client extensions: EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_platform_gbm EGL_KHR_platform_wayland EGL_EXT_platform_wayland
  display extensions: EGL_WL_bind_wayland_display EGL_KHR_partial_update EGL_KHR_config_attribs EGL_KHR_image EGL_KHR_image_base EGL_KHR_fence_sync EGL_KHR_wait_sync EGL_KHR_gl_colorspace EGL_KHR_get_all_proc_addresses EGL_IMG_context_priority EGL_KHR_no_config_context EGL_EXT_image_dma_buf_import EGL_EXT_pixel_format_float EGL_ARM_pixmap_multisample_discard EGL_ARM_implicit_external_sync EGL_KHR_gl_texture_2D_image EGL_KHR_gl_renderbuffer_image EGL_KHR_create_context EGL_KHR_surfaceless_context EGL_KHR_gl_texture_cubemap_image EGL_EXT_image_gl_colorspace EGL_EXT_create_context_robustness
===================================
===================================
OpenGL ES information:
  version: "OpenGL ES 3.2 v1.r25p0-01eac0.9771aca0686ac609bad539142d5c7fce"
  shading language version: "OpenGL ES GLSL ES 3.20"
  vendor: "ARM"
  renderer: "Mali-G52"
  extensions: "GL_ARM_rgba8 GL_ARM_mali_shader_binary GL_OES_depth24 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_EXT_read_format_bgra GL_OES_compressed_paletted_texture GL_OES_compressed_ETC1_RGB8_texture GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_required_internalformat GL_OES_vertex_array_object GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_EXT_texture_rg GL_EXT_texture_type_2_10_10_10_REV GL_OES_fbo_render_mipmap GL_OES_element_index_uint GL_EXT_shadow_samplers GL_OES_texture_compression_astc GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc_hdr GL_KHR_texture_compression_astc_sliced_3d GL_EXT_texture_compression_astc_decode_mode GL_EXT_texture_compression_astc_decode_mode_rgb9e5 GL_KHR_debug GL_EXT_occlusion_query_boolean GL_EXT_disjoint_timer_query GL_EXT_blend_minmax GL_EXT_discard_framebuffer GL_OES_get_program_binary GL_OES_texture_3D GL_EXT_texture_storage GL_EXT_multisampled_render_to_texture GL_EXT_multisampled_render_to_texture2 GL_OES_surfaceless_context GL_OES_texture_stencil8 GL_EXT_shader_pixel_local_storage GL_ARM_shader_framebuffer_fetch GL_ARM_shader_framebuffer_fetch_depth_stencil GL_ARM_mali_program_binary GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_texture_sRGB_decode GL_EXT_texture_sRGB_R8 GL_EXT_texture_sRGB_RG8 GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_OES_texture_storage_multisample_2d_array GL_OES_shader_image_atomic GL_EXT_robustness GL_EXT_draw_buffers_indexed GL_OES_draw_buffers_indexed GL_EXT_texture_border_clamp GL_OES_texture_border_clamp GL_EXT_texture_cube_map_array GL_OES_texture_cube_map_array GL_OES_sample_variables GL_OES_sample_shading GL_OES_shader_multisample_interpolation GL_EXT_shader_io_blocks GL_OES_shader_io_blocks GL_EXT_tessellation_shader GL_OES_tessellation_shader GL_EXT_primitive_bounding_box GL_OES_primitive_bounding_box GL_EXT_geometry_shader GL_OES_geometry_shader GL_ANDROID_extension_pack_es31a GL_EXT_gpu_shader5 GL_OES_gpu_shader5 GL_EXT_texture_buffer GL_OES_texture_buffer GL_EXT_copy_image GL_OES_copy_image GL_EXT_shader_non_constant_global_initializers GL_EXT_color_buffer_half_float GL_EXT_color_buffer_float GL_EXT_YUV_target GL_OVR_multiview GL_OVR_multiview2 GL_OVR_multiview_multisampled_render_to_texture GL_KHR_robustness GL_KHR_robust_buffer_access_behavior GL_EXT_draw_elements_base_vertex GL_OES_draw_elements_base_vertex GL_EXT_buffer_storage GL_EXT_texture_filter_anisotropic GL_ARM_texture_unnormalized_coordinates "
===================================
window.c: Connector: 164 16 1
window.c: Connected
window.c: use preferred mode
window.c: found linked encoder: 163
window.c: found crtcs linked to encoder: 93
window.c: INFO: display has non-square pixels. Non-square-pixels are not supported by flutter. H=203 V=202 r=2.105263
display mode:
  resolution: 480 x 854
  refresh rate: 59.987086Hz
  physical size: 60mm x 107mm
  flutter device pixel ratio: 2.105263
  pixel format: (any)
pluginregistry.c: Initialized plugins: services, text input, raw keyboard plugin, gstreamer video_player, audioplayers,
window.c: Flutter requested render surface before supplying surface dimensions.
gl_renderer.c: Choosing EGL config with pixel format ARGB 8:8:8:8...
egl_gbm_render_surface.c: using fourcc AR24 (ARGB 8:8:8:8) with modifier 0x0
flutter: SocketClientNotifier: connecting to 63fcc43cc0dfb50b8607bb2a.local`
```

TeunVR, 2024-04-23:
modetest output

```
odroid@odroid-teun:~$ modetest
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...done
Encoders:
id	crtc	type	possible crtcs	possible clones
161	0	Virtual	0x00000003	0x00000001
163	93	DSI	0x00000002	0x00000002

Connectors:
id	encoder	status		name		size (mm)	modes	encoders
164	163	connected	DSI-1          	60x107		1	163
  modes:
	index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot
  #0 480x854 59.99 480 482 492 494 854 866 868 928 27500 flags: nhsync, nvsync; type: preferred, driver
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0
	5 link-status:
		flags: enum
		enums: Good=0 Bad=1
		value: 0
	6 non-desktop:
		flags: immutable range
		values: 0 1
		value: 0
	4 TILE:
		flags: immutable blob
		blobs:

		value:
	51 brightness:
		flags: range
		values: 0 100
		value: 50
	52 contrast:
		flags: range
		values: 0 100
		value: 50
	55 saturation:
		flags: range
		values: 0 100
		value: 50
	56 hue:
		flags: range
		values: 0 100
		value: 50

CRTCs:
id	fb	pos	size
[... 643 lines cut ...]
		value: 0
	152 INPUT_HEIGHT:
		flags: immutable range
		values: 0 2304
		value: 0
	153 OUTPUT_WIDTH:
		flags: immutable range
		values: 0 4096
		value: 0
	154 OUTPUT_HEIGHT:
		flags: immutable range
		values: 0 2304
		value: 0
	155 SCALE_RATE:
		flags: immutable range
		values: 4 4
		value: 0
	156 colorkey:
		flags: range
		values: 0 2164260863
		value: 0

Frame buffers:
id	size	pitch
```

ardera, 2024-04-24:
I see you've already added logging to ensure the output selection works correctly.

Can you try the instructions here and send me the log? https://github.com/ardera/flutter-pi/wiki/debugging-output-issues

TeunVR, 2024-04-24:
I made two logs. 
The mipi log is the log where i don't see any output.
The hdmi log is working ok. 

[dmesg-mipi.log](https://github.com/ardera/flutter-pi/files/15095607/dmesg-mipi.log) 
[dmesg-hdmi.log](https://github.com/ardera/flutter-pi/files/15095608/dmesg-hdmi.log)

The drm_info and modetest are attached to the first post.

ardera, 2024-04-24:
the logs look normal, nothing unusual

does the console cursor also keep blinking? or does it stop blinking while flutter-pi is running?

Can you do a `cat /sys/kernel/debug/dri/*/state` while flutter-pi is running and send me the output? (After it should have presented the first frame, so wait a few secs)

TeunVR, 2024-04-25:
Here are two states, for dri/0 and dri/128.
The cursor keeps blinking, i can still use the terminal.

[dri-128-state.log](https://github.com/ardera/flutter-pi/files/15106475/dri-128-state.log)
[dri-0-state.log](https://github.com/ardera/flutter-pi/files/15106477/dri-0-state.log)

TeunVR, 2024-04-25:
One thing i noticed when comparing the dri-states of the MIPI with the dri-state of the HDMI display is that on the HDMI display there is one crtc in the state which is enabled and active. When using the MIPI display there are two crtc's of which the first is not enabled and not active. I don't know if this is expected, just something i noticed when comparing the files 🙂
[image]

The (working) hdmi dsi states:
[dri-128-state-hdmi.log](https://github.com/ardera/flutter-pi/files/15111233/dri-128-state-hdmi.log)
[dri-0-state-hdmi.log](https://github.com/ardera/flutter-pi/files/15111231/dri-0-state-hdmi.log)

TeunVR, 2024-04-25:
Also here are the two dri-state logs for kmscube on MIPI-DSI which is working.
Here i also see a difference in the crtc states.

[kmscube-dri-128-state-mipi.log](https://github.com/ardera/flutter-pi/files/15111725/kmscube-dri-128-state-mipi.log)
[kmscube-dri-0-state-mipi.log](https://github.com/ardera/flutter-pi/files/15111726/kmscube-dri-0-state-mipi.log)

ardera, 2024-04-25:
> Also here are the two dri-state logs for kmscube on MIPI-DSI which is working. Here i also see a difference in the crtc states. [...]

Interesting, though switching HDMI & DSI involves some dtoverlay stuff right? That could maybe affect the available CRTCs.

In the dri state, it seems like with DSI, flutter-pi is there, but fbcon is also still there and showing in front of flutter-pi (because plane ids are used to determine occlusion order in case zpos is equal AFAIK) and flutter-pi just never removes it from the display.

Can you try with [fix/plane-allocation](https://github.com/ardera/flutter-pi/tree/fix/plane-allocation)?

TeunVR, 2024-04-25:
Yes i change dtoverlays to switch between HDMI and DSI. HDMI is disabled when DSI is used.

With the fix i get a very short flash of the display and then back to console mode. Also when i exit the flutter app i get a small flash again. The cursor keeps flashing in this case also.

Here is the dri state output:
[dri-128-state-shortflash.log](https://github.com/ardera/flutter-pi/files/15112385/dri-128-state-shortflash.log)
[dri-0-state-shortflash.log](https://github.com/ardera/flutter-pi/files/15112386/dri-0-state-shortflash.log)

I have also made a modetest log with the hdmi and dsi display to compare. The modetest DSI shows different values for the possible crtcs than the modetest HDMI output. Also the first CRTC in the DSI output has a size of 0x0?
[modetest-hdmi.log](https://github.com/ardera/flutter-pi/files/15112431/modetest-hdmi.log)
[modetest-mipi.log](https://github.com/ardera/flutter-pi/files/15112432/modetest-mipi.log)

ardera, 2024-04-25:
Okay apparently rockchip (or upstream) weston has specific patches for the rockchip driver: https://github.com/JeffyCN/weston/commit/aae3c532036ba9dd5369201303d559535d4f42d9

Seems like there's some stuff the rockchip driver doesn't like, and flutter-pi is doing that (not assigning the N-th plane to the N-th CRTC). I think kmscube is using legacy modesetting by default and that's why it doesn't run into this problem. You can maybe try if `kmscube -A` works.

ardera, 2024-04-25:
Can you try with the latest commit?

TeunVR, 2024-04-25:
kmscube -A gives no eglDupNativeFenceFDANDROID.

The latest commit gives some compile errors here:
```
/home/odroid/flutter-pi-master/src/modesetting.c: In function ‘drmdev_new_from_interface_fd’:
/home/odroid/flutter-pi-master/src/modesetting.c:1031:79: error: suggest parentheses around comparison in operand of ‘&’ [-Werror=parentheses]
 1031 |             if (drmdev->planes[i].possible_crtcs & (1 << primary_plane_index) != 0) {
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
In file included from /home/odroid/flutter-pi-master/src/modesetting.c:23:
/home/odroid/flutter-pi-master/src/util/logging.h:15:55: error: format ‘%d’ expects a matching ‘int’ argument [-Werror=format=]
   15 |     #define LOG_DEBUG(fmtstring, ...) fprintf(stderr, "%s: " fmtstring, __FILE__, ##__VA_ARGS__)
      |                                                       ^~~~~~
/home/odroid/flutter-pi-master/src/modesetting.c:1034:17: note: in expansion of macro ‘LOG_DEBUG’
 1034 |                 LOG_DEBUG("Primary plane %d does not support CRTC %d.\n", primary_plane_index);
      |                 ^~~~~~~~~
/home/odroid/flutter-pi-master/src/modesetting.c:1034:68: note: format string is defined here
 1034 |                 LOG_DEBUG("Primary plane %d does not support CRTC %d.\n", primary_plane_index);
      |                                                                   ~^
      |                                                                    |
      |                                                                    int
/home/odroid/flutter-pi-master/src/modesetting.c:1039:78: error: suggest parentheses around comparison in operand of ‘&’ [-Werror=parentheses]
 1039 |             if (drmdev->planes[i].possible_crtcs & (1 << cursor_plane_index) != 0) {
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
In file included from /home/odroid/flutter-pi-master/src/modesetting.c:23:
/home/odroid/flutter-pi-master/src/util/logging.h:15:55: error: format ‘%d’ expects a matching ‘int’ argument [-Werror=format=]
   15 |     #define LOG_DEBUG(fmtstring, ...) fprintf(stderr, "%s: " fmtstring, __FILE__, ##__VA_ARGS__)
      |                                                       ^~~~~~
/home/odroid/flutter-pi-master/src/modesetting.c:1042:17: note: in expansion of macro ‘LOG_DEBUG’
 1042 |                 LOG_DEBUG("Cursor plane %d does not support CRTC %d.\n", cursor_plane_index);
      |                 ^~~~~~~~~
/home/odroid/flutter-pi-master/src/modesetting.c:1042:67: note: format string is defined here
 1042 |                 LOG_DEBUG("Cursor plane %d does not support CRTC %d.\n", cursor_plane_index);
      |                                                                  ~^
      |                                                                   |
      |                                                                   int
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/flutterpi_module.dir/build.make:115: CMakeFiles/flutterpi_module.dir/src/modesetting.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/flutterpi_module.dir/all] Error 2
```

TeunVR, 2024-04-25:
I fixed the compile-errors and i've got visuals now 👍 Great!

ardera, 2024-04-25:
> The latest commit gives some compile errors here:

Oh sry about that, I wrote that patch in like 5s because I had to leave

> I fixed the compile-errors and i've got visuals now 👍 Great!

Nice! I also pushed the proper version now.

ardera, 2024-04-25:
fixed by #414

closed 2024-04-25 by ardera