ardera/flutter-pi #413: error: Couldn't create GBM surface for rendering - Rasberry pi 5 with 7 inch official screen

7aku8, 2024-04-24:
Hey, I'm kinda noob in embedded programming so sorry if that's a trivial issue.
I've been struggling for few hours, trying to run Flutter app with `flutter-pi` on Raspberry pi 5 [4gb and Debian GNU/Linux 12 (bookworm)] with official touch screen display.

I skipped 4. step from "Configuring your Raspberry PI" as I wasn't able to find this setting in `raspi-config`. I also wasn't able to find GL driver settings so I enabled fake KMS manually by changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` in `config.txt` file.

Display itself works fine.

I built an app with this command: ```bash
flutterpi_tool build --arch=arm64 --release
```

And then tried to run it with command: ```bash
flutter-pi --release /home/pi/ui
```

And I ended up with that error: ```
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card0". Continuing.
flutter-pi.c: Device "/dev/dri/card2" doesn't have a display connected. Skipping.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
modesetting.c: Couldn't add buffer as DRM fb. drmModeAddFB2WithModifiers: Invalid argument
egl_gbm_render_surface.c: Couldn't add GBM buffer as DRM framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
```

I tried several different config.txt changes without any success. Setting option `display_auto_detect` to `0` disables whole screen.
What I can add is that it used to work just fine with HDMI display.

Do you guys have any ideas whether that's a library issue or I'm doing something wrong?

ardera, 2024-04-25:
Can you try without the `fkms` change? Could be that no longer works on Pi 5, though it shouldn't be necessary anymore anyway.

7aku8, 2024-04-25:
Sure, so I reinstalled OS to make sure that other changes I did will not affect it. I went through the setup process once again, skipping `fkms` change. Finally I ran into exactly same issue as before.

[raspinfo output](https://paste.debian.net/hidden/d3018e47/)

7aku8, 2024-04-30:
@ardera FYI I tried to use Pi4 with 64-bit Raspbian Bullseye instead of Pi5 and it works perfectly fine.

closed 2024-04-30 by ardera

anirudh-walia, 2024-05-02:
running it on a rpi5 with a 3.5 inch tft display 
getting this same error 
```
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card0". Continuing.
flutter-pi.c: Device "/dev/dri/card1" doesn't have a display connected. Skipping.
Segmentation fault

```

ardera, 2024-05-02:
@anirudh-walia That's not the same error. How is your TFT display connected to the Pi? What model?

mhnowak, 2024-09-25:
Hi @ardera! I have similar issues like the ones above:

```bash
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card0". Continuing.
flutter-pi.c: Device "/dev/dri/card2" doesn't have a display connected. Skipping.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
```

I didn't worry about it earlier, but I noticed I cannot change colors in the app, so for example if I run the app with such code:

```dart
Scaffold(
      backgroundColor: Colors.red,
      body: ..
)
```

the background color stays the same default (white or whitish). 

I also skipped these 2 steps on the Raspberry Pi configuration steps, as I couldn't find them in the raspi-config (I thought maybe they're configured like so by default? Idk, I'm a bit new to this)
```markdown
3. You can skip this if you're on Raspberry Pi 4 with Raspbian Bullseye
Enable the V3D graphics driver:
Advanced Options -> GL Driver -> GL (Fake KMS)

4. Configure the GPU memory Performance Options -> GPU Memory and enter 64.
```

I feel like the errors, especially the ones to the `surface.c` might be related to lack of these configuration changes. Is there any other way to set those? Like you mentioned above some configuration file. 

Possibly related to: https://github.com/ardera/flutter-pi/issues/443#issue-2520929739