ardera/flutter-pi #415: Missing Depens in Readme ?

DerKleinePunk, 2024-04-26:
is Package  libvulkan-dev and libseat-dev Needed or not ?
CMake warning ist gone after in install the Libs.

ardera, 2024-04-26:
If you install them you get vulkan support in flutter-pi and session switching support. Though most people don't care about those so they're not really necessary.

DerKleinePunk, 2024-06-13:
Has some an Hint for me ?
After enable libseat i can start the app only with sudo. I don't find where the rigths are Configuriert.

skela, 2025-01-27:
is there anyway to completely exclude vulkan from flutter_pi build or run?

we don't as far as i know install these packages in our docker image, but when we try to run flutter_pi from the generated flutter_assets folder, it exists complaining it cannot find libVulkan.so.

ardera, 2025-03-20:
> is there anyway to completely exclude vulkan from flutter_pi build or run?
> 
> we don't as far as i know install these packages in our docker image, but when we try to run flutter_pi from the generated flutter_assets folder, it exits complaining it cannot find libVulkan.so.

The binary that flutterpi_tool uses is basically a "one-size-fits-all", with most features enabled.
If you want to exclude some features, you can build your own flutter-pi binary and use that to run the bundle