ardera/flutter-pi #417: Different resolution in flutter-pi than that of the OS (DSI panel)

DonatelloX, 2024-05-02:
I have a DSI panel on Raspberry CM4 with last kernel 6.6.28. The resolution of the DSI panel is 1364x768pixel.
In the terminal I have no problem:
```
fbset 
mode "1364x768"
    geometry 1364 768 1364 768 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/0
endmode
```
the resolution seems correct.

When I run my application the resolution from flutter is only 733x413pixel. So the application does not display correctly.
I tried with _videomode_ or _dimensions_ options but no change, the resolution in flutter enviroment is always 733x413pixel.
The _-r 90_ option works by rotating the graphics but does not fix the resolution size.

What could be the problem in the resolution?
Thank you very much

ardera, 2024-05-03:
> When I run my application the resolution from flutter is only 733x413pixel.

That's probably the resolution in logical pixels, if you multiply that with the device pixel ratio you should get the actual size in device pixels.

flutter-pi assumes, if no explicit `--dimensions` argument is provided, that any connected DSI display is the Raspberry Pi 7 inch screen. I'll change that so that's only assumed if the resolution is also 800x480.

Though you said you tried with `--dimensions` as well, does that still not work? I'm using `--dimensions` pretty often and it works for me. Keep in mind the argument following it should be the physical dimensions of the screen in millimeters, e.g. for the pi 7 inch screen it would be: `--dimensions 155,86`

DonatelloX, 2024-05-03:
Yes, I used `SystemChrome.setPreferredOrientations` but it has no effect.

closed 2024-05-13 by DonatelloX