ardera/flutter-pi #419: Installing on iMX6ULL

rimb05, 2024-05-07:
Hi, 

Amazing project!  I got my flutter apps working successfully on RPi, and now I'm trying to get it working on my iMX6ULL demo board.  The proc / cpuinfo output is:

processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 109.09
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5
Hardware        : Freescale i.MX6 Ultralite (Device Tree)
Revision        : 0000

In your opinion, what's the best way to start porting flutter-pi to this system?  Should I first cross compile flutter-pi, or try to build it on-device?  I've got a fresh buildroot image built, but I can also use Ubunto Core or Debian if needed if you think one of those distros would have more built-in support/libraries, etc.  I would love to get your guidance on how to proceed.

Thanks very much!

rimb05, 2024-05-07:
I decided to compile it on my Zero 2W and just copy the executable over to the target board (similar architectures).  Now I get this error when I try to launch a flutter application:

flutter-pi.c: Could not query DRM device list: No such file or directory

I assume this is related to the DRI driver?  My /dev/dri is empty on this device.  I have an LCD panel attached through the RGB interface and I have disabled wayland.

Any idea what my next steps should be?  I read that flutter-pi needs a fake KMS driver?  How would I install this?

I appreciate the help.

ardera, 2024-05-08:
Hey, 

thanks for the kind words.

iMX6ULL is generally not the best hardware to run flutter-pi on, I have to say. First problem is that it doesn't have a GPU. That is technically not a requirement for flutter, as it can use software rendering as well. But that path is currently not implemented in flutter-pi.

The other problem is that it doesn't have a KMS driver, only an fbdev one (IIRC). That's also not supported by flutter-pi right now.

I have an iMX6Q here lying around, I'll look into getting it working with software rendering and fbdev (someone on discord tried to use it on iMX6ULL as well), but I can't give a timeframe unfortunately

ardera, 2024-05-08:
In the meantime, I think you could try using etnaviv.

Etnaviv is normally a replacement for the closed-source Vivante graphics driver (which is not present on the ULL), but IIRC it also comes with a KMS driver. And then use llvmpipe as the graphics driver (that's basically OpenGL emulated using software). That sounds like it would be slower than flutters internal software rendering, but some people from the flutter team hinted on discord it might actually be faster; so I think it's worth a try.

rimb05, 2024-05-08:
Wow, it would certainly be amazing to have broader support in flutter-pi for GPU-less chips.  I'm happy to help test..

rimb05, 2024-05-08:
Is the #custom-embedders channel invite-only?  I couldn't find it.

rimb05, 2024-05-08:
In any case, thank you for what you've done here.  Seeing my flutter application come alive on a pi was simply amazing and made my day.  I hope I can make it work on my processor.