ardera/flutter-pi #420: Audioplayers MissingPluginException

Winterteq, 2024-05-17:
Hey!

flutter-pi for the most part has worked great for my app. it seems to be running (albeit a little slow because of my old RPI3!).

However I am using the audioplayers package on my app - it works fine when I run on my windows machine, but when I build on WSL and move the bundle across to my rpi and run with flutter-pi I get the following error:

`[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method create on channel xyz.luan/audioplayers)
#0 MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:332:7)
<asynchronous suspension>
#1 AudioplayersPlatform.create (package:audioplayers_platform_interface/src/audioplayers_platform.dart:18.5)
<asynchronous suspension>
#2 AudioPlayer._create (package:audioplayers/src/audioplayer.dart:161:7)
<asynchronous suspension>`

To confirm the steps I have taken is:

1. Developed functioning app on Windows dev machine using libraries that all support linux.
2. Used WSL with latest flutter stable build to build the code using `flutter build bundle`
3. Transferred the assets from the "build" folder in my project over to my raspberry pi
4. used "flutter-pi build <project directory>

The app runs and appears on the screen, but the features requiring the audioplayers app are throwing this error.

any help would be appreciated!

HSVDO, 2024-05-30:
Hi, 

I ran into this problem, too.

In addition to the points mentioned above:

I followed the instructions given in the [readme](https://github.com/ardera/flutter-pi/blob/master/README.md#audioplayers) and verified the audio setup using `gst-launch-1.0 -v playbin uri=file:///home/pi/my_apps_flutter_assets/assets/path_to_audio.mp3`.

Using `gst-launch` everything works fine and the audiofiles are played as expected.

I tried both `audioplayers` versions: `^5.0.0` and `^6.0.0` with the same error.

Any ideas on how to fix this would be appreciated :)

popgas-admin, 2025-03-24:
Any updates on this topic? Same error here

ardera, 2025-03-24:
Did you build flutter-pi with audioplayers support?
```
cmake -DBUILD_GSTREAMER_AUDIO_PLAYER_PLUGIN=ON -DTRY_BUILD_GSTREAMER_AUDIO_PLAYER_PLUGIN=OFF ...
```

philseeley, 2025-11-20:
Hi, did anyone get a solution to this one?

I'm running the latest version of flutter-pi on Linux:
<pre>
$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 13 (trixie)"
NAME="Debian GNU/Linux"
VERSION_ID="13"
VERSION="13 (trixie)"
VERSION_CODENAME=trixie
DEBIAN_VERSION_FULL=13.1
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
</pre>

I've build with all the BUILD_GSTREAMER... options but I still get the error:
<pre>
MissingPluginException(No implementation found for method listen on channel xyz.luan/audioplayers.global/events)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:365)
<asynchronous suspension>
#1      EventChannel.receiveBroadcastStream.<anonymous closure> (package:flutter/src/services/platform_channel.dart:712)
<asynchronous suspension>
</pre>
The flutter-pi binary appears to have linked with all the correct libraries:

$ ldd /usr/bin/flutter-pi 
	linux-vdso.so.1 (0x0000007f9769f000)
	libgbm.so.1 => /lib/aarch64-linux-gnu/libgbm.so.1 (0x0000007f97620000)
	libsystemd.so.0 => /lib/aarch64-linux-gnu/libsystemd.so.0 (0x0000007f974f0000)
	libinput.so.10 => /lib/aarch64-linux-gnu/libinput.so.10 (0x0000007f97470000)
	libxkbcommon.so.0 => /lib/aarch64-linux-gnu/libxkbcommon.so.0 (0x0000007f97400000)
	libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007f973a0000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f972f0000)
	libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000007f972b0000)
	libGLESv2.so.2 => /lib/aarch64-linux-gnu/libGLESv2.so.2 (0x0000007f97260000)
	libseat.so.1 => /lib/aarch64-linux-gnu/libseat.so.1 (0x0000007f97230000)
	**libgstapp-1.0.so.0 => /lib/aarch64-linux-gnu/libgstapp-1.0.so.0 (0x0000007f971f0000)**
	**libgstallocators-1.0.so.0 => /lib/aarch64-linux-gnu/libgstallocators-1.0.so.0 (0x0000007f971c0000)**
	libdrm.so.2 => /lib/aarch64-linux-gnu/libdrm.so.2 (0x0000007f97180000)
	**libgstvideo-1.0.so.0 => /lib/aarch64-linux-gnu/libgstvideo-1.0.so.0 (0x0000007f97080000)**
	**libgstbase-1.0.so.0 => /lib/aarch64-linux-gnu/libgstbase-1.0.so.0 (0x0000007f96fd0000)**
	**libgstreamer-1.0.so.0 => /lib/aarch64-linux-gnu/libgstreamer-1.0.so.0 (0x0000007f96e30000)**
	libgobject-2.0.so.0 => /lib/aarch64-linux-gnu/libgobject-2.0.so.0 (0x0000007f96da0000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007f96c10000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f96a50000)
	libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000007f96a00000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f97650000)
	libcap.so.2 => /lib/aarch64-linux-gnu/libcap.so.2 (0x0000007f969d0000)
	libmtdev.so.1 => /lib/aarch64-linux-gnu/libmtdev.so.1 (0x0000007f969a0000)
	libevdev.so.2 => /lib/aarch64-linux-gnu/libevdev.so.2 (0x0000007f96960000)
	libwacom.so.9 => /lib/aarch64-linux-gnu/libwacom.so.9 (0x0000007f96930000)
	libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000007f967a0000)
	liborc-0.4.so.0 => /lib/aarch64-linux-gnu/liborc-0.4.so.0 (0x0000007f966d0000)
	libgmodule-2.0.so.0 => /lib/aarch64-linux-gnu/libgmodule-2.0.so.0 (0x0000007f966a0000)
	libffi.so.8 => /lib/aarch64-linux-gnu/libffi.so.8 (0x0000007f96670000)
	libatomic.so.1 => /lib/aarch64-linux-gnu/libatomic.so.1 (0x0000007f96640000)
	libpcre2-8.so.0 => /lib/aarch64-linux-gnu/libpcre2-8.so.0 (0x0000007f96580000)
	libgudev-1.0.so.0 => /lib/aarch64-linux-gnu/libgudev-1.0.so.0 (0x0000007f96550000)
$

ardera, 2025-11-20:
@philseeley Hey, can you try building with these parameters:
```
cmake ... -DBUILD_GSTREAMER_AUDIO_PLAYER_PLUGIN=ON -DTRY_BUILD_GSTREAMER_AUDIO_PLAYER_PLUGIN=OFF
```

The `off` on the second one is important, otherwise it'll happily continue building flutter-pi without the plugin, if required libraries were not found :)

philseeley, 2025-11-22:
Look like this is related to:

- https://github.com/bluefireteam/audioplayers/issues/1154
- https://github.com/ardera/flutter-pi/issues/408
- https://github.com/ardera/flutter-pi/issues/120 - looks like it previously assumed Linux and worked

Using a test app I can see that the when running under flutter-pi the, possibly updated, way of accessing platform plugins now shows an "Unknown" platform.

Steps to recreate test app:
<pre>
$ flutter create --template plugin --platforms linux test
$ cd test/example/
$ flutter build linux
</pre>
Running the test app on the Pi4 with X forwarding to my laptop:
<pre>
$ build/linux/arm64/release/bundle/test_example
</pre>
Gives:

[image]

Running under flutter-pi:
<pre>
$ flutterpi_tool build --arch=arm64 --cpu=pi4 --release 
$ ~/flutter-pi/build/flutter-pi --release build/flutter-pi/pi4-64/
</pre>
Gives:

[image]

Hope that helps.