ardera/flutter-pi #421: Orange Pi Zero 3 configuration

jupiter9595, 2024-05-17:
As my predecessors said, great work! Thank you for your effort!

I've successfully configured flutter-pi on Rasperry Pi Zero 2W

Now I would like to run it on Orange Pi Zero 3

I'm running Ubuntu 22.04 compiled by OrangePi 

Successfully ran GUI test:

sudo glmark2
=======================================================
    glmark2 2021.02
=======================================================
    OpenGL Information
    GL_VENDOR:     Mesa
    GL_RENDERER:   Mali-G31 (Panfrost)
    GL_VERSION:    3.1 Mesa 23.2.1-1ubuntu3.1~22.04.2
    
But encountered the following errors when running flutter-pi (code recompiled for  aarch64 with > flutterpi_tool build --arch=arm64)

flutter-pi timer_linux2a
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
flutter: The Dart VM service is listening on http://127.0.0.1:41759/hq1w0dqu8fk=/
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.
window.c: Could not commit frame request.
modesetting.c: Commit requested, but drmdev is paused right now.

Has anyone successfully configured Orange Pi Zero 3 (or other OrangePi) board?

ardera, 2024-06-04:
Is there another app running that is using display output?
Can you try running with `sudo`?

jupiter9595, 2024-08-26:
Temporary I don't have Orange Pi Zero 3 board. But what I recall is that I was using sudo and no other app was running. Should this run ok on this board? If yes I will get this board again. It is much more powerful then Raspberry Pi Zero W (including Zero 2W). I got rid of it because other compatibility issues but it will be worth to get it back if flutter-pi works on it.

jupiter9595, 2024-09-15:
@ardera - I've got Orange Pi Zero 3 board again.

My current set up is:

Image used Orangepizero3_1.0.2_ubuntu_jammy_server_linux6.1.31 - for 1GB memory board

Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy

Compiled glmark2-drm (Jeff Geerling guide) - and it shows correctly the test images directly to the screen

Warning: DRM_CAP_ASYNC_PAGE_FLIP not supported, falling back to 'mailbox' mode for SwapInterval(0).
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      Mesa
    GL_RENDERER:    Mali-G31 (Panfrost)
    GL_VERSION:     3.1 Mesa 23.2.1-1ubuntu3.1~22.04.2
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   1920x1080 fullscreen
=======================================================
[build] use-vbo=false: FPS: 153 FrameTime: 6.567 ms
[build] use-vbo=true: FPS: 153 FrameTime: 6.554 ms
[texture] texture-filter=nearest: FPS: 309 FrameTime: 3.242 ms
[texture] texture-filter=linear: FPS: 308 FrameTime: 3.251 ms
[texture] texture-filter=mipmap: FPS: 314 FrameTime: 3.188 ms
[shading] shading=gouraud: FPS: 103 FrameTime: 9.748 ms
[shading] shading=blinn-phong-inf: FPS: 103 FrameTime: 9.740 ms

**However** again, flutter-pi fails. Now with different error messages:

orangepi@orangepizero3:~/myFlutter$ flutter-pi timer_linux2a
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
flutter-pi.c: Error: Could not load flutter engine from any location. Make sure you have installed the engine binaries.
orangepi@orangepizero3:~/myFlutter$ sudo flutter-pi timer_linux2a
[sudo] password for orangepi:
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
flutter-pi.c: Error: Could not load flutter engine from any location. Make sure you have installed the engine binaries.

Can you guide me what to test further?

ardera, 2024-09-15:
How did you build the `timer_linux2a`? Using flutterpi_tool?

Can you do a `ls -lah` inside the `timer_linux2a` dir?

jupiter9595, 2024-09-16:
Yes I was using flutterpi_tool and I can successfully run on Raspberry Pi zero 2W

this is the output:
 ls -lah
total 88M
drwxrwxr-x 6 orangepi orangepi 4.0K May 14 08:15 .
drwxrwxr-x 4 orangepi orangepi 4.0K Sep 14 13:44 ..
-rw-rw-r-- 1 orangepi orangepi  117 May 14 08:12 AssetManifest.bin
-rw-rw-r-- 1 orangepi orangepi  109 May 14 08:12 AssetManifest.json
drwx------ 3 orangepi orangepi 4.0K May 14 08:15 flutter_engine
-rw-rw-r-- 1 orangepi orangepi  208 May 14 08:12 FontManifest.json
drwxrwxr-x 2 orangepi orangepi 4.0K May 14 08:12 fonts
-rw-rw-r-- 1 orangepi orangepi 799K May 14 08:09 icudtl.dat
-rw-rw-r-- 1 orangepi orangepi 9.9M May 14 08:12 isolate_snapshot_data
-rw-rw-r-- 1 orangepi orangepi  35M May 14 08:12 kernel_blob.bin
-rw-rw-r-- 1 orangepi orangepi   32 May 14 08:09 .last_build_id
-rw-rw-r-- 1 orangepi orangepi  43M May 14 08:09 libflutter_engine.so
-rw-rw-r-- 1 orangepi orangepi  78K May 14 08:12 NOTICES.Z
drwxrwxr-x 3 orangepi orangepi 4.0K May 14 08:12 packages
drwxrwxr-x 2 orangepi orangepi 4.0K May 14 08:12 shaders
-rw-rw-r-- 1 orangepi orangepi  13K May 14 08:12 vm_snapshot_data

is anything missing in rsync 'ed folder to OrangePi Zero 3 ?

ardera, 2024-09-16:
Can you run `uname -a` and `file timer_linux2a/libflutter_engine.so`?

jupiter9595, 2024-09-16:
uname -a
Linux orangepizero3 6.1.31-sun50iw9 #1.0.2 SMP Mon Sep 25 13:54:37 CST 2023 aarch64 aarch64 aarch64 GNU/Linux

file timer_linux2a/libflutter_engine.so
timer_linux2a/libflutter_engine.so: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=ce82832dd87f067e60787d28e8789cc3137c0f37, not stripped

ardera, 2024-09-16:
Your Orange Pi 4 uses arm64, but the flutter app is built for arm32. Try running flutterpi_tool with `--arch=arm64`

jupiter9595, 2024-09-17:
Sorry, my bad, I should not have used Raspberry Pi 2W 32bit version with OrangePi Zero 3 Ubuntu22.04 64bit version

It took a while to update my Virtual Ubuntu 22.04 AMD64 env and upgrade flutter and flutterpi_tool version

Now it works fine --->

 flutter-pi timer_linux2a_arm64
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
flutter: The Dart VM service is listening on http://127.0.0.1:33131/wR9yfp5iZLg=/

I can see and use app GUI.

From this command line output; is there anything I should further check / update?

Thank you for your help!

PS: Next, I am going to verify Radxa Zero 3 E board GUI driver Mali-G52 r1 (Panfrost) with Armbian 24.5 OS  Hope this will be useful people visiting your repository.

ardera, 2024-09-17:
nice! yeah it's definitely useful to have a list of known-working boards. Ideally there'd even be automated testing on those boards. Otherwise it's possible I make a change and it breaks without my knowledge. The linux KMS API is a bit brittle in that regard.

> flutter-pi timer_linux2a_arm64
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
flutter: The Dart VM service is listening on http://127.0.0.1:33131/wR9yfp5iZLg=/
> 
> I can see and use app GUI.
> 
> From this command line output; is there anything I should further check / update?

Not really, all the messages there are just informational. They could indicate a problem if something was not working, but if everything is working, it's all fine. But I'll improve that in the next update, in my current working branch I've changed it so that information is only logged when something is actually wrong.

> PS: Next, I am going to verify Radxa Zero 3 E board GUI driver Mali-G52 r1 (Panfrost) with Armbian 24.5 OS Hope this will be useful people visiting your repository.

Yep definitely, thanks!