ardera/flutter-pi #433: Memory leak on NXP IMX8x

0xkelvin, 2024-07-13:
- Memory Usage
  + Startup done: 11% of 773 MB
  + Executed 2 hours: ~50% of 773 MB (high chance memory leak issue)
  
  
[image]

Please let me know what should i provide more to address this issue

ardera, 2024-07-14:
Yep looks like a memory leak, what kind of packages & plugins are you using? Theoretically this could also come from memory allocated using `dart:ffi` not being freed (but could also definitely be a memory leak in flutter-pi).

Is there otherwise anything special about your setup that would be good to know when I try to reproduce? Do you have a mouse connected, is there anything interesting in the flutter-pi log, etc.

ardera, 2024-07-15:
Can you try this:
- build with `cmake ... -DENABLE_ASAN=On`
- running your app with flutter-pi until you think some memory has leaked
- quit your flutter app sanely using [`SystemNavigator.pop`](https://api.flutter.dev/flutter/services/SystemNavigator/pop.html)
- send me the complete commandline output

0xkelvin, 2024-08-04:
hi, i am going to close this issue, since it is due to our implementation on flutter, not related to flutter-pi

closed 2024-08-04 by 0xkelvin