ardera/flutter-pi #436: Directory Structure after build

jebos, 2024-08-02:
I was playing around with flutter-pi and I found that the flutter-pi on my target assumes a directory structure like

app/data/flutter-assets
app/lib/

while when building the resulting bundle on my machine creates a structure where everything is placed in flutter-assets.

Did I miss to read the correct documentation?
Is this a bug?
Do I miss a step, or do I run incompatible versions of flutter-pi?

-> when i copy the files manually to the expected locations it works as expected.

Thanks alot.

ardera, 2024-08-02:
How did you build flutter-pi? The directory structure you describe is supported by flutter-pi, but it's not the default. It only gets activated when configuring with `-DFILESYSTEM_LAYOUT=meta-flutter`, which is used when building flutter-pi for yocto with https://github.com/meta-flutter/meta-flutter.git.

jebos, 2024-08-03:
Thanks. Yes I did build a yocto with meta flutter.

closed 2024-08-03 by jebos