ardera/flutter-pi #437: Flutter-pi compiled app on CoreMP183 errors

mgrouch, 2024-08-02:
When I run it 

````
root@CoreMP135:~# flutter-pi --release ./boatinstrument
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
WARNING: Detected llvmpipe (ie. software rendering) as the OpenGL ES renderer.
         Check that flutter-pi has permission to use the 3D graphics hardware,
         or try running it as root.
         This warning will probably result in a "failed to set mode" error
         later on in the initialization.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Function not implemented
egl_gbm_render_surface.c: Will retry without modifiers

````

When I start moving mouse:

````
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
````

I do not see mouse pointer.

Related:  https://github.com/philseeley/boatinstrument/issues/3

Thanks

ardera, 2024-08-16:
Sorry for the delay. That's probably a bug. The hardware mouse cursor is not really well-supported, it's only a help for debugging, if you don't have a touchscreen.

Maybe this helps for now: https://gist.github.com/ardera/e7ddf7c8e477d258a26dfa9c40002626 (though you'll still get the error messages)

I'll keep this issue open until I fix it.

mgrouch, 2024-10-09:
I have touchscreen with USB interface and no mouse

````
root@CoreMP135:~/boatinstrument# ./flutter-pi --release .
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
WARNING: Detected llvmpipe (ie. software rendering) as the OpenGL ES renderer.
         Check that flutter-pi has permission to use the 3D graphics hardware,
         or try running it as root.
         This warning will probably result in a "failed to set mode" error
         later on in the initialization.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Function not implemented
egl_gbm_render_surface.c: Will retry without modifiers
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not provided by any .service files
#0      DBusClient._callMethod (package:dbus/src/dbus_client.dart:1120)
<asynchronous suspension>
#1      DBusClient.callMethod (package:dbus/src/dbus_client.dart:621)
<asynchronous suspension>
#2      WakelockPlusLinuxPlugin.toggle.<anonymous closure> (package:wakelock_plus/src/wakelock_plus_linux_plugin.dart:47)
<asynchronous suspension>
#3      WakelockPlusLinuxPlugin.toggle (package:wakelock_plus/src/wakelock_plus_linux_plugin.dart:40)
<asynchronous suspension>

window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
window.c: GBM BO has unsupported framebuffer stride 128, expected was: 88
````

Errors occur when I move a finger over the touchscreen

Thanks

ardera, 2024-11-01:
Is your touchscreen controller an AR1100 by any chance?

mgrouch, 2024-11-08:
My touchscreen:

````
[  492.416884] input: WingCool Inc. TouchScreen Touchscreen as /devices/platform/soc/5800d000.usbh-ehci/usb2/2-1/2-1.4/2-1.4.2/2-1.4.2:1.0/0003:27C0:0818.000D/input/input13
[  492.418995] hid-generic 0003:27C0:0818.000D: input: USB HID v1.10 Device [WingCool Inc. TouchScreen] on usb-5800d000.usbh-ehci-1.4.2/input0
````

There seems to be some timeouts in Flutter GestureDetector with touch:

There are also these:

````
libinput error: event2  - WingCool Inc. TouchScreen Mouse: client bug: event processing lagging behind by 629ms, your system is too slow
libinput error: event2  - WingCool Inc. TouchScreen Mouse: client bug: event processing lagging behind by 890ms, your system is too slow

````

Thanks