ardera/flutter-pi #438: "io.flutter.rast" received signal SIGSEGV, Segmentation fault on NXP IMX8

0xkelvin, 2024-08-05:
We are using Yocto to build-up an custom linux image for our nxp im8x board. 
And using flutter-pi to run our UI app which develop by flutter version 3.22.2

After compile linux image successfully, we flash the image and power on the board, the flutter UI comes up and work well. 

Then stop the UI app ( press Control + C), power off the board, then turn on again. 
At this stage, we expect the flutter UI shows up and works normally, but we got segment fault issue : 

```
Thread 5 "io.flutter.rast" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xfffff13db090 (LWP 1436)]
0x0000fffff7b2dba0 in ?? () from /usr/lib/libGLESv2.so.2
(gdb) bt
#0  0x0000fffff7b2dba0 in ?? () from /usr/lib/libGLESv2.so.2
#1  0x0000fffff7b364a0 in ?? () from /usr/lib/libGLESv2.so.2
#2  0x0000fffff7b43048 in ?? () from /usr/lib/libGLESv2.so.2
#3  0x0000fffff7b16de0 in ?? () from /usr/lib/libGLESv2.so.2
#4  0x0000fffff7b17498 in ?? () from /usr/lib/libGLESv2.so.2
#5  0x0000fffff7ad69a4 in ?? () from /usr/lib/libGLESv2.so.2
#6  0x0000fffff23646e8 in GrGLOpsRenderPass::onDraw(int, int) () from /usr/share/hmi-app/lib/libflutter_engine.so
#7  0x0000fffff2290068 in GrOpFlushState::drawMesh(GrSimpleMesh const&) () from /usr/share/hmi-app/lib/libflutter_engine.so
#8  0x0000fffff231d04c in GrOp::execute(GrOpFlushState*, SkRect const&) () from /usr/share/hmi-app/lib/libflutter_engine.so
#9  0x0000fffff231ce28 in skgpu::ganesh::OpsTask::onExecute(GrOpFlushState*) () from /usr/share/hmi-app/lib/libflutter_engine.so
#10 0x0000fffff227e350 in GrDrawingManager::executeRenderTasks(GrOpFlushState*) () from /usr/share/hmi-app/lib/libflutter_engine.so
#11 0x0000fffff227de18 in GrDrawingManager::flush(SkSpan<GrSurfaceProxy*>, SkSurfaces::BackendSurfaceAccess, GrFlushInfo const&, skgpu::Mo
#12 0x0000fffff227e7c8 in GrDrawingManager::flushSurfaces(SkSpan<GrSurfaceProxy*>, SkSurfaces::BackendSurfaceAccess, GrFlushInfo const&, o
#13 0x0000fffff209666c in flutter::DlSkCanvasAdapter::Flush() () from /usr/share/hmi-app/lib/libflutter_engine.so
#14 0x0000fffff1fb949c in flutter::EmbedderExternalView::Render(flutter::EmbedderRenderTarget const&, bool) ()
   from /usr/share/hmi-app/lib/libflutter_engine.so
#15 0x0000fffff1fbaab8 in flutter::EmbedderExternalViewEmbedder::SubmitFlutterView(long, GrDirectContext*, std::_fl::shared_ptr<impeller:)
   from /usr/share/hmi-app/lib/libflutter_engine.so
#16 0x0000fffff265b0f4 in flutter::Rasterizer::DrawToSurfaceUnsafe(long, flutter::LayerTree&, float, std::_fl::optional<fml::TimePoint>)
    () from /usr/share/hmi-app/lib/libflutter_engine.so
#17 0x0000fffff265a7f4 in flutter::Rasterizer::DrawToSurfacesUnsafe(flutter::FrameTimingsRecorder&, std::_fl::vector<std::_fl::unique_ptro
#18 0x0000fffff2659064 in flutter::Rasterizer::DrawToSurfaces(flutter::FrameTimingsRecorder&, std::_fl::vector<std::_fl::unique_ptr<flutto
#19 0x0000fffff2659e6c in flutter::Rasterizer::DoDraw(std::_fl::unique_ptr<flutter::FrameTimingsRecorder, std::_fl::default_delete<flutte)
   from /usr/share/hmi-app/lib/libflutter_engine.so
#20 0x0000fffff265cbb8 in std::_fl::__function::__func<flutter::Rasterizer::Draw(std::_fl::shared_ptr<flutter::Pipeline<flutter::FrameIte-
, void (std::_fl::unique_ptr<flutter::FrameItem, std::_fl::default_delete<flutter::FrameItem> >)>::operator()(std::_fl::unique_ptr<flutteo
#21 0x0000fffff265972c in flutter::Pipeline<flutter::FrameItem>::Consume(std::_fl::function<void (std::_fl::unique_ptr<flutter::FrameItemo
#22 0x0000fffff2659350 in flutter::Rasterizer::Draw(std::_fl::shared_ptr<flutter::Pipeline<flutter::FrameItem> > const&) ()
   from /usr/share/hmi-app/lib/libflutter_engine.so
#23 0x0000fffff2674ce0 in std::_fl::__function::__func<fml::internal::CopyableLambda<flutter::Shell::OnAnimatorDraw(std::_fl::shared_ptr<o
#24 0x0000fffff1ff588c in fml::MessageLoopImpl::FlushTasks(fml::FlushType) () from /usr/share/hmi-app/lib/libflutter_engine.so
#25 0x0000fffff1ffbf2c in fml::MessageLoopLinux::Run() () from /usr/share/hmi-app/lib/libflutter_engine.so
#26 0x0000fffff1ff5740 in fml::MessageLoopImpl::DoRun() () from /usr/share/hmi-app/lib/libflutter_engine.so
#27 0x0000fffff1ffb8b8 in std::_fl::__function::__func<fml::Thread::Thread(std::_fl::function<void (fml::Thread::ThreadConfig const&)> coo
#28 0x0000fffff1ffb4e8 in fml::ThreadHandle::ThreadHandle(std::_fl::function<void ()>&&)::$_0::__invoke(void*) ()
   from /usr/share/hmi-app/lib/libflutter_engine.so
#29 0x0000fffff7572f34 in start_thread (arg=0xffffffffe1af) at pthread_create.c:463
#30 0x0000fffff74cb21c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

if we flash the image again, it can start normally at first time. 

we are really appreciate for any help or advices 

thanks

ardera, 2024-08-06:
Looks like a GL driver issue. Bit mysterious that it works the first time but doesn't work the second time though.

One thing flutter does different on second launch is that it loads precompiled GL shaders from disk that were stored there in the first run, instead of compiling from scratch. Maybe the Vivante driver doesn't like that. Can you try launching with `--purge-persistent-cache`?

0xkelvin, 2024-08-23:
hi @ardera 
it works, thanks
i will close this issue.

closed 2024-08-23 by 0xkelvin