ardera/flutter-pi #439: Is't possible to replace `systemd` with `libuv`

HuMoran, 2024-08-15:
Is't possible to replace `systemd` with `libuv`? Using `systemd` will cause the boot to start very slowly.

Here is an example:

https://github.com/sony/flutter-embedded-linux/commit/595f7f5bb02a32eb09a3ad136a5f9736e0fa7244

ardera, 2024-08-15:
`libsystemd` is just an utility library used by systemd, it's not systemd itself. I haven't tried but I'm pretty sure you can use `libsystemd` without `systemd`.

Other than that, yeah, you could also modify it to use `libuv` instead

HuMoran, 2024-08-16:
If using `libsystemd`， the system needs to use `systemd` as the init program. Did I understand it wrong?

ardera, 2024-08-16:
AFAICT, no:

```console
hannes@alpaka-pi:~ $ ldd /lib/aarch64-linux-gnu/libsystemd.so.0
	linux-vdso.so.1 (0x0000007f8c837000)
	libcap.so.2 => /lib/aarch64-linux-gnu/libcap.so.2 (0x0000007f8c6d0000)
	libgcrypt.so.20 => /lib/aarch64-linux-gnu/libgcrypt.so.20 (0x0000007f8c5c0000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f8c570000)
	libzstd.so.1 => /lib/aarch64-linux-gnu/libzstd.so.1 (0x0000007f8c4b0000)
	liblz4.so.1 => /lib/aarch64-linux-gnu/liblz4.so.1 (0x0000007f8c460000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f8c2b0000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f8c7fa000)
	libgpg-error.so.0 => /lib/aarch64-linux-gnu/libgpg-error.so.0 (0x0000007f8c260000)
```

https://packages.debian.org/de/sid/libsystemd0

no dependency on the actual systemd (https://packages.debian.org/de/sid/systemd), udev, etc

HuMoran, 2024-08-19:
I use 'buildroot' to build an image, which shows that it must depend on 'systemd'` 😂
[image]

ardera, 2024-08-19:
That seems like an issue on their end then :) Or it's intentional, could be it's very rare to depend on libsystemd without depending on systemd, so they just bundled it together.

ardera, 2024-08-20:
did you open an issue with buildroot? if yes, you can link it here and keep this one open. Just if someone else has the same problem in the future.

HuMoran, 2024-08-26:
Sorry, I don't have a GitLab account, I'm from China, and Gitlab banned the China register. 😭

ardera, 2024-08-26:
@HuMoran opened an issue here: https://gitlab.com/buildroot.org/buildroot/-/issues/30

aduskett, 2024-08-27:
This isn't a buildroot issue. libsystemd is provided by systemd, and there isn't a way to build and install libsystemd from the systemd package. For binary distributions, there are usually multiple binary packages generated from a single source, but that is not possible in Buildroot.

HuMoran, 2024-08-28:
Can add the 'libuv' feature?

ardera, 2024-08-28:
> This isn't a buildroot issue. libsystemd is provided by systemd, and there isn't a way to only build and install libsystemd from the systemd package. For binary distributions, there are usually multiple binary packages generated from a single source, but that is not possible in Buildroot.

Thanks @aduskett, yeah seems like installing only libsystemd is a pretty rare use-case
Though it seems like you can in fact install libsystemd separately using `meson install --tags=libsystemd`. And technically you can build separately using `meson compile libsystemd` (or ninja) as well, however the `meson install` command will still continue to build the whole repo anyway. But idk how easy it is to make buildroot work with that

ardera, 2024-08-28:
> Can add the `libuv` feature?

I don't know how likely is that I will work on it, but if you want to work on it, I'd happily accept it. I'm currently working on a refactor for multi-view and I've been switching to use my own little mt-safe sd-event wrapper there exclusively: https://github.com/ardera/flutter-pi/blob/feat/multiview/src/util/event_loop.c

For libuv you'd probably want to implement the same API using libuv in some `event_loop_libuv.c` file instead, and then compile that instead of `event_loop.c` if some cmake option is set

HuMoran, 2024-08-30:
I'm not sure I can do it. This feature is a big project for me (I haven't written C code for many years). I am learning the relevant code. Is there any design Introduction?

doawoo, 2024-09-09:
I've actually done a bit of work on a much much older version of `flutter_pi` that rips out systemd and replaces the it with a very light weight event loop library:

https://github.com/troglobit/libuev

I'll perhaps try to find some time to port all these changes to the latest version if there's interest in that.

ardera, 2024-09-12:
@HuMoran I have added a bit of comments here: https://github.com/ardera/flutter-pi/blob/feat/multiview/src/util/event_loop.h let me know if that helps

@doawoo that'd be great!

legraps, 2024-09-13:
Interesting, I had the same problem with buildroot as @HuMoran. Solved it for me by removing the dependency of systemd being the init system. It works but of course is super hacky. (didnt dare to suggest it on the buildroot mailing list).

So libuev as alternativ would be very nice!