ardera/flutter-pi #440: How to rotate touch input by 180 degrees without rotating the displayed app?

resilem, 2024-08-15:
Hello,

I am wondering how to rotate touch input by 180 degrees without rotating the displayed app? It does not seem this is supported as a runtime argument. If I pass `flutter-pi` the `--rotation 180` flag both touch input and display are rotated 180 degrees.

If I could just apply the rotation to the touch input, this would allow me to configure the display rotation using the display monitor settings while still having touch input work correctly. If I configure the display monitor to flip it's image vertically (180 degrees) then I find that touch events are off by 180 degrees.

I naively tried achieving this by applying the following patch:
```
diff --git a/src/user_input.c b/src/user_input.c
index 652d7df..bd585ac 100644
--- a/src/user_input.c
+++ b/src/user_input.c
@@ -114,7 +114,7 @@ static inline FlutterPointerEvent make_touch_event(FlutterPointerPhase phase, si
     event.pan_x = 0.0;
     event.pan_y = 0.0;
     event.scale = 0.0;
-    event.rotation = 0.0;
+    event.rotation = 180.0;
 
     return event;
 }
@@ -169,7 +169,7 @@ static inline FlutterPointerEvent make_mouse_event(
     event.pan_x = 0.0;
     event.pan_y = 0.0;
     event.scale = 0.0;
-    event.rotation = 0.0;
+    event.rotation = 180.0;
 
     return event;
 }
@@ -220,7 +220,7 @@ static inline FlutterPointerEvent make_stylus_event(FlutterPointerPhase phase, s
     event.pan_x = 0.0;
     event.pan_y = 0.0;
     event.scale = 0.0;
-    event.rotation = 0.0;
+    event.rotation = 180.0;
 
     return event;
 }

```
But this does not seem to work.

~ Maxwell

ardera, 2024-08-15:
This happens sometimes, the best way to fix this is using a udev rule for that particular touchscreen, see: https://github.com/ardera/flutter-pi/issues/164#issuecomment-841879291

resilem, 2024-08-15:
Thanks for the suggestion @ardera worked like a charm.

closed 2024-08-15 by resilem