ardera/flutter-pi #442: Video player init problem

Krakert, 2024-09-10:
I have a problem with loading a video with `gstreamer ` and `video_player`.

code:
```flutter
class PreparationScreen extends StatelessWidget {
  constPreparationScreen({super.key});

  @override
  Widget build(BuildContext context) {
    return const Scaffold(
      backgroundColor: Color(0xFF302E2F),
      body: Center(
        child: BannerPlayer(),
      ),
    );
  }
}

class BannerPlayer extends StatefulWidget {
  const BannerPlayer({super.key});

  @override
  State<BannerPlayer> createState() => _BannerPlayerState();
}

class _BannerPlayerState extends State<BannerPlayer> {
  late VideoPlayerController _controller;

  @override
  void initState() {
    super.initState();
    initializePlayer();

    _controller.setLooping(true);

  }

  Future<void> initializePlayer() async {
    _controller =
        VideoPlayerController.asset('assets/videos/banner.mp4');

    await Future.wait([
      _controller.initialize(),
    ]);
  }

  @override
  void dispose() {
    _controller.dispose();
    super.dispose();
  }

  @override
  Widget build(BuildContext context) {
    _controller.play();
    return AspectRatio(
        aspectRatio: _controller.value.aspectRatio,
        child: Stack(
          alignment: Alignment.bottomCenter,
          children: <Widget>[
            VideoPlayer(_controller),
            VideoProgressIndicator(_controller, allowScrubbing: true),
          ],
        ));
  }
}
```

Just a simple widget that loads a video from the assets. Works on web and Android, but on flutter-pi:

```bash
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
flutter: The Dart VM service is listening on http://127.0.0.1:42009/XFhgcS8bO_U=/
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: UnimplementedError: init() has not been implemented.
#0      VideoPlayerPlatform.init (package:video_player_platform_interface/video_player_platform_interface.dart:43:5)
#1      _videoPlayerPlatform (package:video_player/video_player.dart:34:21)
#2      VideoPlayerController.initialize (package:video_player/video_player.dart:440:25)
#3      _BannerPlayerState.initializePlayer (package:fresh_brew_ng/beverage/beverage_preparation_screen.dart:45:19)
#4      _BannerPlayerState.initState (package:fresh_brew_ng/beverage/beverage_preparation_screen.dart:34:5)
#5      StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5748:55)
#6      ComponentElement.mount (package:flutter/src/widgets/framework.dart:5593:5)
#7      Element.inflateWidget (package:flutter/src/widgets/framework.dart:4468:16)
#8      Element.updateChild (package:flutter/src/widgets/framework.dart:3963:18)
#9      SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6900:14)
#10     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4468:16)
#11     Element.updateChild (package:flutter/src/widgets/framework.dart:3963:18)
#12     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5642:16)
#13     Element.rebuild (package:flutter/src/widgets/framework.dart:5333:7)
#14     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5599:5)
#15     ComponentElement.mount (package:flutter/src/widgets/framework.dart:5593:5)
#16     Element.inflateWidget (package:flutter/src/widgets/framework.dart:4468:16)
#17     Element.updateChild (package:flutter/src/widgets/framework.dart:3963:18)
#18     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5642:16)
#19     Element.rebuild (package:flutter/src/widgets/framework.dart:5333:7)
#20     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5599:5)
...
```

What a I doing wrong here?

Krakert, 2024-09-11:
I was able to get the video to run with `flutterpi_gstreamer_video_player: ^0.1.1+1` But warnings shows and after a while it crashes:

```bash
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
flutter: The Dart VM service is listening on http://127.0.0.1:46339/ugpJgQjzxDc=/
flutter: platformName :- Linux
plugins/gstreamer_video_player/player.c: Failed to set custom HTTP headers because gstreamer source element has no 'extra-headers' property.

(flutter-pi:2906): GStreamer-CRITICAL **: 08:47:40.451: gst_sample_set_caps: assertion 'gst_sample_is_writable (sample)' failed

** (flutter-pi:2906): WARNING **: 08:47:49.412: v4l2h264dec0: 1 frames 451-451 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug

(flutter-pi:2906): GStreamer-CRITICAL **: 08:47:49.463: gst_sample_set_caps: assertion 'gst_sample_is_writable (sample)' failed

** (flutter-pi:2906): WARNING **: 08:49:19.453: v4l2h264dec0: 1 frames 2711-2711 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug
plugins/gstreamer_video_player/player.c: gstreamer error: code: 1, domain: gst-stream-error-quark, msg: Internal data stream error. (debug info: ../gst/isomp4/qtdemux.c(6937): gst_qtdemux_loop (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstDecodeBin:decodebin0/GstQTDemux:qtdemux0:
streaming stopped, reason error (-5))
plugins/gstreamer_video_player/player.c: last gstreamer state change failed. gst_element_get_state(element name: pipeline0): GST_STATE_CHANGE_FAILURE
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(nativeerror, Input/output error, 5, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
<asynchronous suspension>
#2      FlutterpiVideoPlayer.getPosition (package:flutterpi_gstreamer_video_player/src/platform.dart:155:20)
<asynchronous suspension>
#3      VideoPlayerController._applyPlayPause.<anonymous closure> (package:video_player/video_player.dart:586:41)
<asynchronous suspension>
```
my `main`:

```flutter
void main() {
   // I added this check, so it still works for the web and android version to (debugging)
  if (Platform.isLinux) {
    FlutterpiVideoPlayer.registerWith();
  }

  runApp(const MyApp());
}
```

ardera, 2024-11-10:
Sorry for the delay.

You're on Raspberry Pi, right? Can you try building with `-DCMAKE_BUILD_TYPE=Debug`?

At least the 
```
** (flutter-pi:2906): WARNING **: 08:47:49.412: v4l2h264dec0: 1 frames 451-451 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug
...
** (flutter-pi:2906): WARNING **: 08:49:19.453: v4l2h264dec0: 1 frames 2711-2711 left undrained after CMD_STOP, eos sent too early: bug in decoder -- please file a bug
plugins/gstreamer_video_player/player.c: gstreamer error: code: 1, domain: gst-stream-error-quark, msg: Internal data stream error. (debug info: ../gst/isomp4/qtdemux.c(6937): gst_qtdemux_loop (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstDecodeBin:decodebin0/GstQTDemux:qtdemux0:
streaming stopped, reason error (-5))
plugins/gstreamer_video_player/player.c: last gstreamer state change failed. gst_element_get_state(element name: pipeline0): GST_STATE_CHANGE_FAILURE
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(nativeerror, Input/output error, 5, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
<asynchronous suspension>
#2      FlutterpiVideoPlayer.getPosition (package:flutterpi_gstreamer_video_player/src/platform.dart:155:20)
<asynchronous suspension>
#3      VideoPlayerController._applyPlayPause.<anonymous closure> (package:video_player/video_player.dart:586:41)
<asynchronous suspension>
```
parts look like a gstreamer/kernel issue that has been reported some times now. I have an idea for a workaround but I'm not sure it'll work, unfortunately.

These parts:
```
(flutter-pi:2906): GStreamer-CRITICAL **: 08:47:40.451: gst_sample_set_caps: assertion 'gst_sample_is_writable (sample)' failed
```
I'm also not sure are flutter-pi related, there's not a single place that calls `gst_sample_set_caps` in flutter-pi, though it could be it's called from some inlined function or macro. If you build with `-DCMAKE_BUILD_TYPE=Debug`, it should print more precise information as to where that call comes from.

Krakert, 2024-11-11:
@ardera After running the same code with `flutterpi` build in debug mode this is the trace.
It played the video maybe 2/3 time before failing.

```
pi@raspberrypi:~ $ flutter-pi flutter/video/
==============Locale==============
Flutter locale:
  default: en
  locales: en en.UTF-8 en_GB en.UTF-8 en.UTF-8 en_GB en.UTF-8
===================================
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card0". Continuing.
===================================
EGL information:
  version: 1.4
  vendor: Mesa Project
  client extensions: EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_EXT_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless
  display extensions: EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display
===================================
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 23.2.1-1~bpo12+rpt3"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_generate_mipmap_sRGB GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_debug_label GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_polygon_offset_clamp GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_cube_map_array GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_NV_shader_noperspective_interpolation GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clear_texture GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_texture_mirror_clamp_to_edge GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod GL_MESA_bgra "
===================================
window.c: WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
Use the `-d` commandline parameter to specify the physical dimensions of your display.
display mode:
  resolution: 1280 x 800
  refresh rate: 59.909545Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
  pixel format: (any)
pluginregistry.c: Initialized plugins: services, text input, raw keyboard plugin, gstreamer video_player, audioplayers,
[... cut ...]
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
<asynchronous suspension>
#2      FlutterpiVideoPlayer.getPosition (package:flutterpi_gstreamer_video_player/src/platform.dart:155:20)
<asynchronous suspension>
#3      VideoPlayerController._applyPlayPause.<anonymous closure> (package:video_player/video_player.dart:586:41)
<asynchronous suspension>
```

MaksymTiutiunnyk, 2025-04-28:
Greetings!

Have you found any solution for this issue? I am experiencing a very similar problem on a Radxa CM3 board.

Here is the output when I try to run my app:
```
rock@radxa-cm3-rpi-cm4-io:~$ sudo flutter-pi --release test/flutter_assets/
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
arm_release_ver: g24p0-00eac0, rk_so_ver: 7
[ERROR:flutter/runtime/dart_vm_initializer.cc(40)] Unhandled Exception: UnimplementedError: init() has not been implemented.
#0      VideoPlayerPlatform.init (package:video_player_platform_interface/video_player_platform_interface.dart:43)
#1      _videoPlayerPlatform (package:video_player/video_player.dart:33)
#2      VideoPlayerController.initialize (package:video_player/video_player.dart:439)
#3      _AdvertisementModeScreenState.playVideo (package:aura/advertisement_mode_screen.dart:100)
#4      _AdvertisementModeScreenState.playNext (package:aura/advertisement_mode_screen.dart:71)
#5      _AdvertisementModeScreenState.loadMediaFiles (package:aura/advertisement_mode_screen.dart:50)
<asynchronous suspension>
```

To install GStreamer, I used the following command:
`sudo apt install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-plugins-bad gstreamer1.0-libav gstreamer1.0-alsa`

Then I recompiled flutter-pi:
```
rm -rf build
mkdir build && cd build
cmake ..
make -j`nproc`
```

And installed it again:
`sudo make install`

Have I missed anything?
Thanks in advance for any help!