ardera/flutter-pi #443: flutter-pi -r 180 doesn't rotate the shadows

mhnowak, 2024-09-11:
When I used the -r argument to rotate the screen it rotated all the UI nicely, but it didn't rotate the shadows rendering (look for the floating action button at the bottom right of the screen) - they're rendering just like they'd use to render without the rotate. Is there any way to fix that currently?

The rotated image: 
[image]

Not rotated image (expected behaviour):
[image]

mhnowak, 2024-09-11:
I get this kind of errors after I run the command. Could that impact the shadows output?

```bashrc
modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
flutter-pi.c: Could not create drmdev from device at "/dev/dri/card0". Continuing.
flutter-pi.c: Device "/dev/dri/card2" doesn't have a display connected. Skipping.
egl_gbm_render_surface.c: Couldn't create GBM surface for rendering. gbm_surface_create_with_modifiers: Invalid argument
egl_gbm_render_surface.c: Will retry without modifiers
```

ardera, 2024-09-12:
I've never seen that before, not even sure how that would happen. I'll try reproducing that

As a temporary workaround, you can put your entire app in a `RotatedBox` and rotate it that way

mhnowak, 2024-09-12:
Sure. I'm running the latest Debian 12 for raspberry pi on raspberry pi 5. (It's my first pi, but I've read online there's a lot of differences between the previous versions, or the pi OSs). If there's anything else you need from me just let me know.