ardera/flutter-pi #447: Radxa Zero 3 E configuration

jupiter9595, 2024-09-18:
My setup:

This is the board with Ethernet connection only.

Armbian 24.5 using this image -> Armbian_24.5.1_Radxa-zero3_noble_vendor_6.1.43_gnome_desktop

Set to boot to console with sudo systemctl set-default multi-user.target

@radxa-zero3:~/myFlutter$ glmark2 --size 1024x600
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      Mesa
    GL_RENDERER:    Mali-G52 r1 (Panfrost)
    GL_VERSION:     3.1 Mesa 24.0.9-0ubuntu0.1
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   1024x600 windowed
=======================================================
[build] use-vbo=false: FPS: 196 FrameTime: 5.109 ms
[build] use-vbo=true: FPS: 199 FrameTime: 5.026 ms
[texture] texture-filter=nearest: FPS: 335 FrameTime: 2.993 ms
[texture] texture-filter=linear: FPS: 316 FrameTime: 3.166 ms
[texture] texture-filter=mipmap: FPS: 339 FrameTime: 2.956 ms
[shading] shading=gouraud: FPS: 153 FrameTime: 6.561 ms
[shading] shading=blinn-phong-inf: FPS: 153 FrameTime: 6.577 ms
[shading] shading=phong: FPS: 151 FrameTime: 6.660 ms
[shading] shading=cel: FPS: 153 FrameTime: 6.568 ms
[bump] bump-render=high-poly: FPS: 70 FrameTime: 14.467 ms
[bump] bump-render=normals: FPS: 358 FrameTime: 2.797 ms

Works well;)  Thank you @Ardera for your great tool

[image]
image mainly illustrates the board works with flutter-pi. Image quality is law due to reflections.

ardera, 2024-09-18:
perfect, good to know!

It would also help if you uploaded the drm info for you board(s): https://drmdb.emersion.fr/ That way I (and people developing compositors) can check the capabilities of the display controller hw on that board.

jupiter9595, 2024-09-19:
Sure will do, 
I've closely checked out behavior of both (Orange Pi zero 3 and Radxa Zero 3E) and looks that on Radxa Zero 3E cursor is barley visible. Keyboard works fine. Reason for this might be different Linux versions I used. What do you think?
I might be wrong but on  Radxa Zero 3E , Armbian 24.5 has better graphic driver support (better then Radxa's version of Debian)

Additionally I need to investigate further the nstall OS on onboard eMMC. But for that I need to have Radxa Zero 3 W (wireless version)
Flutter-pi + Flutter app booting from onboard eMMC might be very interesting setup for embedded programming of some kind of machine controlers.