ardera/flutter-pi #454: MissingPluginException (making custom native code)

Krakert, 2024-10-10:
I was adding some native code to mount and list files from a USB drive. I was following the docs: [Add a Linux platform-specific implementation
](https://docs.flutter.dev/platform-integration/platform-channels#step-7-add-a-linux-platform-specific-implementation)

But unfortinaly when ran it says: `No implementation found for method getUSBFiles`

[here you mentioned this](https://github.com/ardera/flutter-pi/issues/166#issuecomment-810584506) but is this still a thing, and can I write native code?

Running the same code under ubuntu via wsl works fine.

ardera, 2024-11-01:
Sorry for the delay. It's a bit misleading, the docs you linked are for building platform-specific code for the linux desktop embedder, but that unfortunately won't work when trying to write native code for flutter-pi or ivi-homescreen. In the flutter-pi case, you have to write the code in flutter-pi itself, for examples see the sentry plugin in `src/plugins`.

Though in general, isn't mounting a USB drive and listing files something that can be done in dart? E.g.
```dart
final result = Process.runSync('mount', ['/dev/sdXY', '/tmp/myUSBXXXX']);
final files = Directory('/tmp/myUSBXXXX').listSync();
```

Krakert, 2024-11-01:
Here's a clearer and typo-corrected version of your message:

Thanks for your input! I couldn’t find a good source for mounting a drive in Dart, which is why I switched to C.

In our use case, the current GUI and main application are split and communicate via MQTT, so the PoC I’m working on also works this way. I have a Python script that lists the files via MQTT, and the Flutter PoC subscribes to it.

closed 2024-11-01 by Krakert