ardera/flutter-pi #455: Failed to load dynamic library

emersonsiega, 2024-10-13:
I'm trying to use a custom native library, lets say "mylib.so". 
I can run the Flutter app and/or build to native without any problem.

After build with flutter-pi and run in a raspberry pi 3 board, the app cannot find the library.

```bash
> flutter-pi --release /home/pi/app
> flutter: Invalid argument(s): Failed to load dynamic library 'mylib.so': mylib.so: cannot open shared file: No such file or directory
```

How can I build and run the app with custom libraries?

*I have tried to move this library to /usr/local/lib and /usr/lib, but didn't solve..

pezi, 2024-11-02:
Check if LD_LIBRARY_PATH is empty for your env?
If yes, 
`export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH`
befor calling flutter_pi.

Levi-Lesches, 2025-04-10:
Hi @emersonsiega, did the above suggestion fix your problem? If so, you can close this issue

closed 2025-05-03 by github-actions[bot]