ardera/flutter-pi #456: modesetting.c: Couldn't move mouse cursor. drmModeMoveCursor: Bad address

vinsfortunato, 2024-10-26:
I'm running a demo counter Flutter app on a BananaPi M2 Berry and it is working great, so I can confirm it working on this board.

However the mouse does not move and I get the following error in console:
`modesetting.c: Couldn't move mouse cursor. drmModeMoveCursor: Bad address`

The mouse is actually rendered on the top-left corner of the screen but rarely moves (most of the cases it stays fixed in place).
This board does not support hardware cursor since it does not have a dedicated cursor plane. I checked it by using `drm_info`.

The funny thing is that when I tried to add software cursor inside the Flutter application using the following code: https://gist.github.com/ardera/e7ddf7c8e477d258a26dfa9c40002626 the cursor now works properly, it is drawn on top of the Flutter software cursor! 

So I'm actually painting a transparent cursor in Flutter and everything seems to work good. 
The error message is being still printed in console however:
`modesetting.c: Couldn't move mouse cursor. drmModeMoveCursor: Bad address`

abuzarra, 2025-04-10:
Hi, 
I'm experiencing a similar issue when running it on an ARM64-based platform. Is there any known workaround, aside from implementing a software cursor directly in the Flutter application?

Thanks,

ardera, 2025-04-11:
That's a bug in the flutter-pi hardware mouse cursor implementation that I have yet to fix, unfortunately. Happens when the graphics driver has some special requirements on the layout of the mouse cursor buffer that flutter-pi doesn't satisfy, e.g. requiring a specific framebuffer stride or specific dimensions.

@abuzarra Yes, I think the software mouse cursor is a good workaround for now.