ardera/flutter-pi #465: Turn off screen after certain time

danielkastberg, 2024-12-25:
I would like to turn off the screen after midnight. Is it possible since flutter pi runs without X11 or Wayland?
I read online but all posts are mentioning something related to X11 or Wayland like "xset", "tvservice" or "xrandr" but no use since they require X11 or Wayland running.

Can this be done or do I have to turn the power off the PI during nighttime?

DisDis, 2024-12-25:
You can take my example of turning the screen on and off on the Pi4 https://github.com/DisDis/dslideshow/tree/master/deb_factory/template_build_arm64/opt/dslideshow/scripts

danielkastberg, 2024-12-25:
Thank you! I'm away from my PI but will test it when I get back. I used tvservice before and got "Can't find display" but then I know it should work for the flutter pi

tryy3, 2024-12-31:
I had a similar case and thought I would share my solution.
I am running a Pi 5 with the new Touch Screen Display 2 and I couldn't get any of the tools that people recommended (tvservice, xrandr etc.) but managed to find that I could change the brightness directly with the backlight class.

Under `/sys/class/backlight/` I was able to find my display and then set the brightness to 0 or max_brightness.
So In my case I can turn it off using `echo 0 | sudo tee /sys/class/backlight/11-0045/brightness` and turn it on with `echo 20 | sudo tee /sys/class/backlight/11-0045/brightness`

In my case I wanted "screensaver" functionality so changing the brightness worked wonderful since the display is on and reactive just no brightness. So I could just make a widget that lives at the top with a GestureDetector to trigger a timer.
Your case might be a bit different though and brightness might not be enough.

FibreFoX, 2025-02-02:
When running on a Raspberry Pi 3B+ with current Debian Bookworm, I would run `kmsblank -c 0` to disable the display (which is connected via HDMI). But sadly this does not work:
```
Connector 0: blank
Failed to set DPMS: -13
```

When flutter-pi is not running, that command works and disables the display as expected (that tool [waits for a keypress to resume](https://github.com/tomba/kmsxx/blob/aaab406251540429522c5ef7808ee049c65a06d2/utils/kmsblank.cpp#L96) the display). I assume because the way flutter-pi works, the command is "clashing" with my flutter app running.

danielkastberg, 2025-02-03:
I solved it with changing the display driver to fkms.
`dtoverlay=vc4-fkms-v3d`
and then I could control the power with vcgencmd
`vcgencmd display_power 0` && `vcgencmd display_power 1`
Not a sustainable solutions since the fkms has been deprecated but works for now.
I'll try `/sys/class/backlight/` and see if that can work with kms-driver

DisDis, 2025-05-01:
@FibreFoX @danielkastberg  I implemented first version. It works fine. 
https://github.com/ardera/flutter-pi/pull/478