ardera/flutter-pi #466: Questions about Plugin Development for flutter-pi

kyle-seongwoo-jun, 2025-01-05:
Thank you for this amazing project!

I deeply appreciate your efforts in creating and maintaining such an incredible project. 

---

Currently, I am interested in writing plugin code for flutter-pi. This includes both writing new plugins or implementing existing plugins from pub.dev for flutter-pi.

However, I noticed that there isn’t detailed documentation about this topic in the README or the Wiki. I have searched and reviewed related GitHub issues and thought it might be helpful to have this information documented.

From my understanding, there seem to be two main approaches for plugin development:

1. **Writing C code in `src/plugins`**  
   I understand this method involves forking the repository, adding the C implementation in the `src/plugins` directory, building the modified flutter-pi binary, and deploying it to the Raspberry Pi.

2. **Using FFI**  
   This approach follows [C interop using dart:ffi](https://dart.dev/interop/c-interop) as documented in Dart’s official guidelines.

I’d like to confirm if my understanding is correct.  

Assuming it is, I have the following additional questions:

1. In the case of the first approach, is it possible to use the `flutterpi_tool` build method (described as the "New Method" in the README), or is this method incompatible?  

2. Is there a way to build C plugin code during the Flutter app build process, without needing to separately build the flutter-pi binary?

3. (If question 2 is possible) Would it also be feasible to implement a system where plugins for flutter-pi are distributed as pub.dev packages, similar to other custom embedders like Tizen or Sony eLinux?

4. If I want to use C++ code for a plugin, is the second approach (FFI) the only viable option?

Thank you for your time and for reading this!

ardera, 2025-01-13:
Hey @kyle-seongwoo-jun,

yes, your assumptions are correct. You can either have it in the flutter-pi source tree directly, or you build it externally somehow and use `dart:ffi`.

> 1. In the case of the first approach, is it possible to use the flutterpi_tool build method (described as the "New Method" in the README), or is this method incompatible?

That is unfortunately incompatible right now.

> 2. Is there a way to build C plugin code during the Flutter app build process, without needing to separately build the flutter-pi binary?

If you're using custom-devices, you can use the `postBuildCommand` to do anything you want, including building shared libraries and bundling those with the app. In vanilla flutter, there's the native assets feature, which can too be used for building shared libs, but that's yet to be implemented for flutter-pi and flutterpi_tool.

> 4. If I want to use C++ code for a plugin, is the second approach (FFI) the only viable option?

You can write C++ plugin code, there's also a sentry plugin and sentry uses C++. You just need to be a bit careful when interfacing the C code. E.g. the flutter-pi headers don't have
```c++
#ifdef __cplusplus
extern "C" {
#endif
```
guards, so you need to put those around the `#include` directives instead (Or you add them in the headers and open a PR :) )

Can I ask what kind of plugin you want to write? In general, I'm very happy to include any plugins in the flutter-pi repo

kyle-seongwoo-jun, 2025-01-13:
@ardera, Thank you for the detailed response; it was incredibly helpful!

The project I am considering involves creating a system that continuously captures microphone input on a Raspberry Pi, recognizes songs, and visualizes the results in real-time.

To achieve this, I have been thinking about migrating the Linux implementation of the [record package](https://github.com/llfbandit/record) for flutter-pi. Additionally, I am considering using [SongRec](https://github.com/marin-m/SongRec) or [vibra](https://github.com/BayernMuller/vibra/) to implement the song recognition functionality as a Flutter plugin.

(By the way, I just realized that the [Linux implementation](https://github.com/llfbandit/record/blob/master/record_linux/lib/record_linux.dart#L228) of the record package doesn’t rely on a native C++ implementation. Instead, it interacts with the `fmedia` binary directly from Dart. So, it seems like I might be able to make it work without writing additional C code.)

Thank you again for your support!

closed 2025-01-13 by kyle-seongwoo-jun