ardera/flutter-pi #470: Unable to activate flutterpi_tool

JerichoCross, 2025-03-20:
I am unable to run the 'flutter pub global activate' command successfully on my ZorinOS laptop with flutter installed. When I try, I get the following error:

`Note: file is pinned to version 7.0.1 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Note: http is pinned to version 1.2.2 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Note: unified_analytics is pinned to version 7.0.1 by flutter_tools from the flutter SDK.
See https://dart.dev/go/sdk-version-pinning for details.

Because flutterpi_tool >=0.2.0 <0.6.0 depends on http ^0.13.6 and flutterpi_tool <0.2.0 depends on file ^6.1.4, flutterpi_tool
  <0.6.0 requires http ^0.13.6 or file ^6.1.4.
And because every version of flutter_tools from sdk depends on both file 7.0.1 and http 1.2.2, flutterpi_tool <0.6.0 is
  incompatible with flutter_tools from sdk.
And because every version of flutterpi_tool depends on flutter_tools from sdk and flutterpi_tool >=0.5.2 depends on
  unified_analytics >=5.8.0 <7.0.0, every version of flutterpi_tool requires unified_analytics >=5.8.0 <7.0.0.
Because every version of flutterpi_tool depends on flutter_tools from sdk which depends on unified_analytics 7.0.1, every version
  of flutterpi_tool requires unified_analytics 7.0.1.
Thus, flutterpi_tool is forbidden.
So, because pub global activate depends on flutterpi_tool any, version solving failed.
`

flutter doctor reports the following:

`Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.29.2, on Zorin OS 17.2 6.8.0-52-generic, locale en_GB.UTF-8)
[✓] Android toolchain - develop for Android devices (Android SDK version 35.0.0-rc3)
[✓] Chrome - develop for the web
[✓] Linux toolchain - develop for Linux desktop
[✓] Android Studio (version 2024.3)
[✓] VS Code (version 1.98.2)
[✓] Connected device (2 available)
[✓] Network resources

• No issues found!
`

I see that similar issues were reported last year but I just tried this today. Is there something that I need to do to get this to work?

ardera, 2025-03-20:
I just released 0.7.0 a few minutes ago, can you try again? This mostly happens when a new Flutter SDK is released (in this case 3.29.0) and I haven't yet updated flutterpi_tool to be compatible to it.

JerichoCross, 2025-03-20:
Many thanks - this is working now.

closed 2025-03-20 by JerichoCross