ardera/flutter-pi #473: Unable to Start Flutter-Pi using Buildroot - filesystem_layout.c: icudtl file not found!`

MatthewJones517, 2025-03-23:
I am attempting to set up flutter-pi using Buildroot. When I try to do the following:

`flutter-pi --release ~/myapp`

I get the following error:
`filesystem_layout.c: icudtl file not found!`

- I am building my app using the "new" method listed in the README.md `flutterpi_tool build --arch=arm64 --cpu=pi4 --release`
- According to the docs, I shouldn't have to install the flutter engine. I did it anyway to try and resolve the issue. 
- The Flutter engine being installed by Buildroot is `3.27.1`
- I build my app using flutter `3.27.1` to match. 
- I have confirmed that the `icudtl.dat` file is in `flutter_assets`
- I get the same error regardless of what the current working directory is when I try to initiate flutter-pi
- I have copied `icudtl.dat` to the home directory and that didn't work. 
- I tried installing the `icu` package in buildroot and that didn't help. 

Weirdly, I initially copied over just my `flutter_assets` folder but it appears that flutter-pi is expecting it to be in `~/my_app/data/flutter_assets`. I had to update my directory structure to match that. 

I recognize that the one thing I haven't tried is manually installing the [flutter engine binaries](https://github.com/ardera/flutter-engine-binaries-for-arm), but again... I'm already doing that via Buildroot AND I'm using the "new" build method on my dev machine (Ubuntu 24) so I shouldn't need it in the first place. 

I did dig through the flutter-pi code a bit and found this:

```
set(FILESYSTEM_LAYOUT "default" CACHE STRING "Where to look for the icudtl.dat, app.so/libapp.so, flutter asset bundle.")
set_property(CACHE FILESYSTEM_LAYOUT PROPERTY STRINGS ${FILESYSTEM_LAYOUTS})
```

So whatever is in FILESYSTEM_LAYOUTS is where it's expecting to find the icudtl.dat? If that's the case, it's already in the flutter_asset_bundle, and it seems to be finding that. 

I did find issue #378. The resolution there appeared to be matching the flutter version to the engine binaries. I already did that (and once again... I'm using the new build method... So...)

I also found issue #383, but again... I'm using the new build system so I shouldn't need the binaries. Also, copying icudtl.dat around didn't fix the problem. 

This is profoundly frustrating and I'd appreciate some insight.

MatthewJones517, 2025-03-23:
**UPDATE:** So copying the icudtl.dat file to ~/my_app/data DID work, but now I'm getting a different error. "/root/my_app/lib/libapp.so" does not exist, but is necessary for release / profile mode. 

This is weird... It's like it's expecting a whole directory structure that straight up doesn't exist if I follow the instructions in the README. 

Once again, any insight would be really appreciated. Right now the only thing in /data is flutter_assets.

ardera, 2025-03-24:
Hey!

Generally, there's 2 different fs layouts: *default / flutter-pi* and *meta-flutter*.

Most people should be using the default layout. Only apps and flutter-pi's that are built using buildroot or meta-flutter are using the `meta-flutter` filesystem layout: https://github.com/buildroot/buildroot/blob/96ed66f75cc92d50e8297a0da1e944085e9c718e/package/flutter-pi/flutter-pi.mk#L27

With the meta-flutter layout (which follows the layout of the linux desktop bundles):
-  the `libapp.so` is at `app-dir/lib/libapp.so`
- `icudtl.dat` is at `app-dir/data/icudtl.dat`, and
- the assets are at `app-dir/data/flutter_assets`.

The precise lookup logic is defined here: https://github.com/ardera/flutter-pi/blob/c7e0ff581b040a3390a47144550c48f8fc275bff/src/filesystem_layout.c#L254-L286

flutterpi_tool does not work that well yet for buildroot/yocto usage. First of all, as you found out as well, `flutterpi_tool build` will build the app in the normal layout, which is not the one that a yocto/buildroot flutter-pi binary on target expects. What could work is using `flutterpi_tool run` directly. That will use a pre-built flutter-pi binary that uses the normal fs layout; but that might also fail because of different library versions. The only other solution is letting buildroot build the app.

MatthewJones517, 2025-03-24:
Hey @ardera! That makes perfect sense. I'm going to look into letting buildroot build the app, but at least I know the root cause now. I appreciate your response. I'm going to close the issue because I feel like I have what i need. :)

closed 2025-03-24 by MatthewJones517