ardera/flutter-pi #475: What's the Best Way to Deal With Native Shared Libaries?

MatthewJones517, 2025-03-30:
I'm building a Flame game. The `flame_audio` package uses AudioPlayer under the hood. As you pointed out in the README, this ends badly. We need a lot of players and the Pi can't handle it. 

I've gotten around this by using `flutter_soloud` for audio. This is awesome, but requires a native shared library to run. 

flutterpi_tool doesn't build this by default. I have to:

- Do a Linux build of the app on my Pi
- Copy out the `libflutter_soloud_plugin.so`
- Run `flutterpi_tool` on my dev machine, 
- Copy `libflutter_soloud_plugin.so` into `flutter_assets`
- Run on the Pi by doing `LD_LIBRARY_PATH=myapp_flutter_assets flutter-pi myapp_flutter_assets`

There has got to be a better way of doing this. Is it possible for `flutterpi_tool` to build the shared library for soloud? Can I run this without overwriting `LD_LIBRARY_PATH`?

ardera, 2025-04-11:
> There has got to be a better way of doing this. Is it possible for flutterpi_tool to build the shared library for soloud? Can I run this without overwriting LD_LIBRARY_PATH?

I'm afraid not :/ It is possible to add support for dart native assets to flutterpi_tool, but adding support for the linux desktop build system basically requires implementing the Linux Desktop Plugin API, which is just not possible because it heavily depends on GTK+.

Though, have you seen the [`AudioPool`] class of flame_audio package? Does that help anything? Or is it still too many players?

MatthewJones517, 2025-04-11:
`AudioPool` is where I started actually. Even using the pools it still overwhelmed the Pi. I beat my head against this for a while before I realized that the issue was that there's only so many `AudioPlayers` the Pi can handle. 

I removed `flutter-pi` and just ran my app in a desktop environment on the Pi. I witnessed the same issues there, which made me realize that a different approach was required. A colleague recommended soloud and it resolved the issues. 

Thanks for the response. I'll start looking at options to work around this for my specific needs. I just didn't want to re-invent the wheel if there was a better option. 

Thanks for all your work on `flutter-pi` and the various RPi packages. I think the community as a whole doesn't realize just how much is possible because of them.

closed 2025-04-11 by MatthewJones517