ardera/flutter-pi #477: Vulkan does not support video.

DisDis, 2025-04-28:
my kernel, RaPi5:
Linux dslideshow1 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

When I execute:
`flutter-pi --vulkan ./flutter_assets/
`
I got:
`Video player: flutter-pi: gl_renderer.c:582: gl_renderer_supports_egl_extension: Assertion `renderer != NULL' failed.`

Video works fine in EGL mode.

DisDis, 2025-04-28:
I found that. flutter-pi does not support video in Vulkan mode :(
https://github.com/ardera/flutter-pi/blob/master/src/plugins/gstreamer_video_player/frame.c#L16

ardera, 2025-04-29:
Yeah unfortunately the flutter engine does not yet support external textures (which are used to display the video frames) when using Vulkan: https://github.com/flutter/flutter/issues/117937

It is technically possible to work around this by using a platform view instead of an external texture, and have flutter-pi manually composite everything together. However that's basically adding a vulkan renderer to flutter-pi. That's definitely possible but seems a bit redundant. I'd really prefer the flutter engine does the rendering, given that it has its own very capable & widely compatible rendering engine, impeller.

eximius313, 2026-10-31:
It seems that https://github.com/flutter/flutter/issues/117937 is finally completed