ardera/flutter-pi #479: Shared Object not found (libmetadata_god.so)

krush62, 2025-05-03:
Thank you very much for this incredible project. Your work makes life much easier. Vielen, vielen Dank!

I created a small media player application that I would like to run on my RaspberrryPi4. The setup went through smoothly (The asset bundle was created using Debian). My project uses the plugin metadata_god. When running the project using flutter-pi --release <asset> an unhandled exception is thrown:

`Failed to load dynamic library 'libmetadata_god.so': libmetadata_god.so: cannot open shared object file: No such file or directory.`

The library itself is present on the Debian machine but absent in the flutter_assets directory. I tried to copy the file from build\native_assets\linux to the assets base directory before copying it to the pi, but without success.

What am I doing wrong?

krush62, 2025-05-03:
Is this probably related to #448 ?

ardera, 2025-05-05:
Hey @krush62 

thanks for the kind words!

Yes, this is indeed the same problem as #448 . Unfortunately linux desktop plugins are not compatible with flutter-pi, because they have their own build system which does not work with flutter-pi/flutterpi_tool. That applies to metadata_god as well