ardera/flutter-pi #495: Display Scaling Issue on Raspberry Pi 5 with MIPI Display

mosynthkey, 2025-09-29:
First of all, thank you for this wonderful project! I'm currently developing a Flutter application that runs on macOS, Windows, and Raspberry Pi 5, and I'm having a great development experience.

I have a question regarding screen size and DPR (Device Pixel Ratio) handling.
On macOS and Windows, widgets are displayed at the expected Flutter widget sizes. However, when I tested on Raspberry Pi 5 with a MIPI display, the DPR is reported as 2.24, causing widgets to appear much larger than intended.
(I'm getting the DPR using mediaQuery.devicePixelRatio)

After examining the source code, I found that the pixel ratio is calculated based on the actual display size using:
```
pixel_ratio = (10.0 * width) / (width_mm * 38.0);
```
Even when I pass the actual screen size using the -d option, I get the same result (DPR of 2.24 with oversized widget display).

What would be the best approach to resolve this issue of oversized widgets?
- Modify flutter_pi to add an option that forces pixel_ratio = 1
- Use Transform on the Flutter app side to scale to the ideal size
- Provide a fake larger display size via the -d option to achieve the ideal widget size

Additionally, could you help me understand what the value 38.0 in the pixel_ratio calculation represents?

Thank you for taking the time to help with this issue.

ardera, 2025-09-29:
Hey,

what's your display size in mm and pixels? 

The formula for device pixel ratio is derived from here: https://api.flutter.dev/flutter/dart-ui/FlutterView/devicePixelRatio.html
38 is the number of logical pixels per cm. (On other words, device pixel ratio is the ratio of your screens dpi to the default dpi of 96 ppi)

mosynthkey, 2025-09-30:
Thank you for the detailed explanation about the device pixel ratio formula.

To answer your question about display size:
- Physical dimensions: 94×151mm
- Resolution: 1280×800 pixels  
- Screen size: 7 inches

I appreciate the reference to the Flutter documentation and the clarification about the 38 logical pixels per cm calculation. This helps me understand how the device pixel ratio is determined based on the screen's DPI relative to the default 96 PPI.

ardera, 2025-09-30:
In that case, you have a dpi of 215, which is 2.24 times the default DPI of 96 :)

I can still add an arg for forcing a pixel ratio of 1.0, if you want.

closed 2026-09-17 by ardera